MarlinFirmware/Marlin #17595: [FR] Homing aligns Z axis with double Z endstops then uses bed probe to set Z=0

mikeshub, 2020-04-18:
### Description

I would like to use the Z endstops on my printer just to align the gantry. They are the optical switch type and they are not very consistent. 

The BL touch works great for a consistent home so that I don't have to mess with baby stepping. I use the BL Touch and two blocks to align the gantry on one of my printers which is fine.

I have a new one that I have installed BL Touch on which works as expected. I would like to use the double Z endstops to eliminate the use two blocks to level the gantry to the frame. Any inaccuracy in the sensors would not be noticable insofar as aligning the gantry. Then the BL Touch would be used for homing to set Z=0.

### Feature Workflow

maybe a define like

`#define G28_HOME_W_PROBE_ALIGN_W_ENDSTOP
`

and require Z_MIN_PROBE_PIN, Z2_USE_ENDSTOP

The regular usage of the printer would be unchagned so there would be no need to modify or add g code commands or arguments. 

1. G8 commanded
2 .Z raise if needed3
3. X and Y are homed.
4. Printer moves head to a position that is safe for the double Z endstops to be triggered.
5. A Z homing cycle is completed using the double Z endstops.
6. Print head rises.
7. Print head moves to homing position for bed probe.
8. Probe deploys / pause for user if needed .
9. A Z homing cycle is completed using the bed probe. This location is defines Z=0 for the printer.

#### Additional Information

A 0.1 - 0.2mm error is nothing for gantry alignment, esp if the gantry is 400mm. Any slant error introduced by inconsistencies would be cleaned up in the mesh generation process. 

0.1 - 0.2mm error makes the first layer adhesion quite frustrating. 

The printer that I use the blocks to level the gantry is very easy to use. Preheat, level, then print. I never have to baby step it. I want this new printer to be the same way without having to use two blocks to level the gantry.

GMagician, 2020-04-19:
Isn't this what G34 already does? simply using bl-touch as "end stop" to align gantry?

mikeshub, 2020-04-19:
@GMagician two reasons. One is that takes quite some time. What I am suggesting is just a slight increase to the time it take to home Z. 

The other thing is I am printing on a cheap 400mm bed that has a fairly even bowl shape, but IMHO it is not good enough for me to align the gantry. The bed has about 1 mm of total variation from the highest edge to the lowest part of the center. It also shifts a bit on me if I have to really scrape a model off the bed. My concern with G34 is that the large warped bed is going to introduce skew errors. 

I'll mess with it some today and if it ends up working well enough I will report back. I guess since it would only really warrant being run once it could be run with the leveling process if it works well.

GMagician, 2020-04-19:
> My concern with G34 is that the large warped bed is going to introduce skew errors

That will be probably true... G34 start with an "at least" planar bed as reference.

ManuelMcLure, 2020-04-19:
It doesn't really matter if one side of the bed is lower than the other for ``G34`` - what you're really looking for is for the gantry alignment to be repeatable so that the mesh offsets apply.

ManuelMcLure, 2020-04-21:
I don't know why it would take 5 minutes to level with ``G34`` - in my case it only takes about 2 minutes for the worst-case scenario of 5 iterations (and that's with double probing each point) and usually it only does 1 or 2 iterations to get to the 0.02 so it takes less than a minute. Given that a print usually takes hours, I don't consider a few minutes at the start of the print to be an issue. Heating the bed takes a _lot_ longer than even a 5 iteration ``G34`` does.

mikeshub, 2020-04-21:
5 minutes was just a number pulled out of the air. Main point being two Z home cycles is going to be faster and far more consistent because it will be independent of any bed misalignment issues.

ManuelMcLure, 2020-04-21:
But using ``G34`` ensures that the axis is exactly aligned to the bed every time since you're measuring the actual bed surface. The end stops could get out of alignment with the actual bed.

mikeshub, 2020-04-21:
I already explained that to you. The bed being out of alignment is the point of mesh leveling. If I want to maximize the dimensional accuracy of the printer the gantry has to have the same tilt each time. 0.1 - 0.2mm at over 400 mm distance is nothing. 0.1-0.2mm for z spacing means the difference in a 0.1mm first layer working or not.

Coding up a kludge for it was trivial and it works great. There is however a significant philosophical question of what's the most right way to do this and I assure you my current method is not it.

mikeshub, 2020-04-22:
*crickets*

I'll make a PR in 2.1 and see if this generates anymore interest there.

gsart84, 2020-12-26:
I would also like to have this function for the reasons mentioned above.
The difference is that my machine has no hardware limit switches but has sensorless homing.
The auto align G34 function requires more time than the approach of the limit switches. 

I imagine it like this:

1: Home X2 (and stay there)
2: Home X1 (and stay there)
3: Home Y 
4: Home Z (bump Z1 and Z2) 
5: Raise Z (for me here 15mm)
6: Move head to a normal Z probe position
7: Probe Z with bl-touch

sargonphin, 2023-06-01:
Any update on this feature? I am in the same situation as @mikeshub here, with the same Chiron bowl-shaped-bed-slinger and I would love to have this feature for also the same reason, at the difference that I'm using an inductive sensor for Z=0. Inaccuracies introduced by the optical sensors and very random bed warping makes first layer almost never a done-on-first-try