MarlinFirmware/Marlin #17596: Marlin disable heaters before end of print.

dobryj, 2020-04-18:
### Bug Description

In short: command to disable heaters (M104) is executed intermediately during parse of input out of order of g-code. Other printing code for extrusion is executed in order. Result: Marlin disable heaters before end of print and is not able to finish last layer.

Problem is reproducible with input of G-code from external output. (Serial/USB, for ex octoprint).
I don't know if code from SD card have same behavior.
 
### My Configurations

Marlin 2.0.5.3, board SKR PRO v1.1, MCU STM32F407ZGT6

This MCU have lot of RAM, therefore I can extend buffers. 

`#define MAX_CMD_SIZE 128`
`#define BUFSIZE 64`
`#define RX_BUFFER_SIZE 32768`

Full configuration in attachment

[marlinCfg.zip](https://github.com/MarlinFirmware/Marlin/files/4497484/marlinCfg.zip)

### Steps to Reproduce

1. print gcode from external serial/USB line
2. On the last layer watch temperature setting
3. Heater is disabled before end of last layer

**Expected behavior:** Execute M104 in order in g-code or create in-order and out-of-order variants of this command.

**Actual behavior:** M104 is executed too early before end of print.

#### Additional Information

AnHardt, 2020-04-18:
`  #define BLOCK_BUFFER_SIZE 256`
is far larger than useful. Decrease to maybe 64 or 32.

The  other way is to send all moves, then M400. After that switch off the heaters.

dobryj, 2020-04-18:
Decrease BLOCK_BUFFER_SIZE is not solution when last few momements are realy long.
M400 don't have any documentation related to M104. Frankly spoken I am not able to find any documentation about in order and out of order gcode execution.
Motivation for this behaviour is clear, but where is it documented? Include solution.

dobryj, 2020-04-18:
Is "G4 P0" equivalent to "M400" from this point of view?

dobryj, 2020-04-18:
I found sufficient documentation here:
https://www.reprap.org/wiki/G-code#Buffering

Closed.

dobryj, 2020-04-18:
It is not bug. It is feature.
https://sd.keepcalms.com/i/keep-calm-its-not-a-bug-its-a-feature.png

closed 2020-04-18 by dobryj