MarlinFirmware/Marlin #17597: [BUG] Cannot Compile Archim 2 on Arduino IDE

AFprinter, 2020-04-18:
After the latest bugfix release I can no longer compile marlin 2.0 in Arduino IDE.  It's telling me that I have to un-comment TMC_USE_SW_SPI and it is.  

Actual error message:
"Archim2 requires Software SPI. Enable TMC_USE_SW_SPI in Configuration_adv.h."

### Steps to Reproduce

1. Compile latest bugfix in Arduino IDE
2. Failure

[Configuration and _adv.zip](https://github.com/MarlinFirmware/Marlin/files/4497524/Configuration.and._adv.zip)

ellensp, 2020-04-21:
you have all stepper drivers set to and invalid option "TMC2103" 
eg #define X_DRIVER_TYPE  TMC2103
It should be #define X_DRIVER_TYPE  TMC2130

With this wrong the firmware doesn't think you have SPI capable stepper drivers and is getting confused.

AFprinter, 2020-04-21:
Well then.. thank you very much. I feel really really dumb right now

closed 2020-04-21 by AFprinter