MarlinFirmware/Marlin #17598: [BUG] Mega 2560 bootloader partially rewrited by firmware

furry-cheetah, 2020-04-18:
Hello. 

I find an intresting bug. If firmware use almost all memory of mcu (99%) it flash correctly but only one time. All next times after compiling and uploading everytime have avrdude verification error and mcu wont boot. 
Compiling and uploading Marlin 2.0.5 from Atom and Visual studio with platform IO.
When it come, i using USBASP programmer to write bootloader from Arduino IDE, then platform IO to new upload.

First time i think this is the end of memory (bad blocks) on genuine arduino and buy a chinese clone. Then upload firmware, test and modify some settings in source code, second upload finish with the same verification error like on first board. I think - something wrong with upload. My experiment finally find a problem - too big firmware partially rewrite bootloader and board go partially bricked. Board restore working after flashing a new bootloader and accept new upload of firmware. If firmware use ~97% of controller memory and previous upload with succes - upload correctly everytime, before you exceed size and touch bootloader. All next uploads with smaller size wont upload too, before restoration bootloader.

I think need a add to marlin firmware check of available mcu memory, if firmware greater than allowed area - show message and stop compiling.

I have a arduino mega+ramps+tmc2130 SPI+full graphic controller with sd on custom delta style printer and too difficult everytime partially teardown printer and board sandwich to able connect ICSP port.

Thanks for a great firmware!

furry-cheetah, 2020-04-18:
>  If firmware use ~97% of controller memory and previous upload with succes - upload correctly everytime, before you exceed size and touch bootloader.

Correction: Previous uploads not exceed of 97%

furry-cheetah, 2020-04-19:
Sometime flashing with 99% memory usage go fine few times at row, but if initialize eeprom have a the same problem. Next flashing finish with verification error until reflash bootloader.

ellensp, 2020-04-19:
Did you protect the bootloader by setting the fuses?
The mega is normally configured with a 8k bootloader that is protected form any user writes by the fuses. 
```
   "efuse": "0xFD",
    "hfuse": "0xD8",
    "lock_bits": "0x0F",
    "lfuse": "0xFF",
    "unlock_bits": "0x3F"
```
ie http://eleccelerator.com/fusecalc/fusecalc.php?chip=atmega2560&LOW=FF&HIGH=D8&EXTENDED=FD&LOCKBIT=0F

furry-cheetah, 2020-04-19:
Nope. I just like noob upload bootloader from arduino ide, not from avrstudio or avrdude. Arduino ide configure fuses automatically. I know about 8k bootloader and 256k mega2560 flash size. 99% of firmware is about 251-252k memory. I think it partially touch bootloader. 

You maybe right, firmware flashes fine, but partially flash to protected bootloader zone and verification go wrong. Its strange, but it repair with reflashing bootloader. Need a more experiments.

I see the same issues from other peoples, but i find solution - just reflash bootloader if have a verification error when flashing firmware. 

I think need add to firmware comparsion of size final rom and free space of chip. 

I forced to utilize maximum memory of mega 2560 before i go to 32bit, my printer use almost all options. (Delta, tmc2130 spi, graphic lcd, sd card, bltouch (ubl calibration), bluetooth, light and power supply control) Before installation of a TMC drives i use Repetier firmware, but it have a much less functionality with this drives and have problem with bltouch. Now i migrate to Marlin and almost everything setup, but in final find this bug with bootloader. 

[image]

furry-cheetah, 2020-04-19:
**Thats a output of console from Atom:**
```
Linking .pio\build\mega2560\firmware.elf
Checking size .pio\build\mega2560\firmware.elf
Building .pio\build\mega2560\firmware.hex
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=======   ]  73.4% (used 6012 bytes from 8192 bytes)
Flash: [==========]  99.4% (used 252528 bytes from 253952 bytes)
Configuring upload protocol...
AVAILABLE: wiring
CURRENT: upload_protocol = wiring
Looking for upload port...
Auto-detected: COM5
Uploading .pio\build\mega2560\firmware.hex
 
avrdude: AVR device initialized and ready to accept instructions
 
Reading | ################################################## | 100% 0.01s
 
avrdude: Device signature = 0x1e9801
avrdude: reading input file ".pio\build\mega2560\firmware.hex"
avrdude: writing flash (252528 bytes):
 
Writing | ################################################## | 100% 36.27s

avrdude: 252528 bytes of flash written
avrdude: verifying flash memory against .pio\build\mega2560\firmware.hex:
avrdude: load data flash data from input file .pio\build\mega2560\firmware.hex:
avrdude: input file .pio\build\mega2560\firmware.hex contains 252528 bytes
avrdude: reading on-chip flash data:
 
Reading | ################################################## | 100% 28.25s
 
avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x3c000
         0x0f != 0x03
avrdude: verification error; content mismatch
 
avrdude: safemode: Fuses OK
 
avrdude done.  Thank you.
 
*** [upload] Error 1
========================= [FAILED] Took 198.19 seconds =========================

```

**Then i upload bootloader from Arduino IDE:**
```

         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
[... 75 lines cut ...]
avrdude: Device signature = 0x1e9801
avrdude: reading input file ".pio\build\mega2560\firmware.hex"
avrdude: writing flash (252528 bytes):
 
Writing | ################################################## | 100% 36.27s

avrdude: 252528 bytes of flash written
avrdude: verifying flash memory against .pio\build\mega2560\firmware.hex:
avrdude: load data flash data from input file .pio\build\mega2560\firmware.hex:
avrdude: input file .pio\build\mega2560\firmware.hex contains 252528 bytes
avrdude: reading on-chip flash data:
 
Reading | ################################################## | 100% 28.23s
 
avrdude: verifying ...
avrdude: 252528 bytes of flash verified
 
avrdude: safemode: Fuses OK
 
avrdude done.  Thank you.
 
========================= [SUCCESS] Took 78.97 seconds =========================
```

Same code first time have error with upload (prewious time i think big firmware something do with bootloader zone), then i update bootloader and upload firmware again - and voila! Its method works.

ellensp, 2020-04-19:
bootloader is at 3E000 (256k - 8k)
error is at 0x3c000 (256k-16k)

In your writing bootloader log you have
avrdude: 261406 bytes of flash written 
Which is very large for a bootloader!
check your file "C:\Program Files (x86)\Arduino\hardware\arduino\avr/bootloaders/stk500v2/stk500boot_v2_mega2560.hex"
It should be 21017 bytes on disk. 
Also it doesn't show setting H or L  or E fuses like it should...

I suspect this is a Upload using programmer log, not a burn bootloader log.

ellensp, 2020-04-19:
after checking a bunch of stuff. I think the issue is a bug in the standard boot loader.
Standard bootloader on most atmega2560 is  
https://github.com/arduino/Arduino-stk500v2-bootloader/blob/master/stk500boot.c#L275
In this code is 
```
#if FLASHEND > 0x0F000
	#define BOOTSIZE 8192
#else
	#define BOOTSIZE 2048
#endif

#define APP_END  (FLASHEND -(2*BOOTSIZE) + 1) 
```
FLASHEND is 0x3FFFF on mega2560's so the the BOOTSIZE is set to 8192, but the APP_END is set to 0x3FFFF - (2*0x2000) + 1  = 0x3BFFE 

The code then uses "if (eraseAddress < APP_END )" so uploading over serial will only erase bytes from 0x00000 to 0x3C000. So 0x3C000--0x3E000 is not erased and can't be reused till a full erase is done, such as uploading a new bootloader. 

I also notice that Prusa noticed this with their custom boot loader for the einsy rambo and removed the *2. see here.
https://github.com/prusa3d/stk500v2-prusa/commit/708b9bf1ecd068993965de74eb17f6e1b046bef7

Moving forward... I think you need a better boot loader.

furry-cheetah, 2020-04-19:
> It should be 21017 bytes on disk.

Loooks ok. 21017 bytes  (24576 on disc) This bootloader provide with arduino ide 1.8.12 from arduino.cc
[image]

I read fuses with usbasp programmer: 
`C:\Program Files (x86)\Arduino\hardware\tools\avr\bin>avrdude.exe -c usbasp -p ATmega2560 -U lfuse:r:-:i -v -C ..\etc\avrdude.conf`

```
avrdude.exe: Version 6.3-20190619
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2014 Joerg Wunsch

             System wide configuration file is "..\etc\avrdude.conf"

             Using Port                    : usb
             Using Programmer              : usbasp
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             serial program mode           : yes
             parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             ByteDelay                     : 0
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
               flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
               lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
               signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

             Programmer Type : usbasp
             Description     : USBasp, http://www.fischl.de/usbasp/

avrdude.exe: auto set sck period (because given equals null)
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: safemode: lfuse reads as FF
avrdude.exe: safemode: hfuse reads as D8
avrdude.exe: safemode: efuse reads as FD
avrdude.exe: reading lfuse memory:

Reading | ################################################## | 100% 0.02s

avrdude.exe: writing output file "<stdout>"
:01000000FF00
:00000001FF

avrdude.exe: safemode: lfuse reads as FF
avrdude.exe: safemode: hfuse reads as D8
avrdude.exe: safemode: efuse reads as FD
avrdude.exe: safemode: Fuses OK (E:FD, H:D8, L:FF)

avrdude.exe done.  Thank you.
```

Looks similar like you proved:

>    "efuse": "0xFD",
>     "hfuse": "0xD8",
>     "lock_bits": "0x0F",
>     "lfuse": "0xFF",
>     "unlock_bits": "0x3F"

Sorry, i am not a programmer, i am a electronics engeneer, but with programming skills :) I may write something wrong and do something wrong, but i try to do everything to find a problem. Thats intrested to me and peoples, who have the same prblem or maybe have in furure :)

ellensp, 2020-04-19:
yip, ignore that post re sizes,and see last post above.. I change my mind as more information came to light.

ellensp, 2020-04-19:
try optiboot for mega2560  https://github.com/MCUdude/MegaCore/blob/master/avr/bootloaders/optiboot_flash/bootloaders/atmega2560/16000000L/optiboot_flash_atmega2560_UART0_115200_16000000L_B7_BIGBOOT.hex

furry-cheetah, 2020-04-19:
Ok.

Upload optiboot bootloader (optiboot_flash_atmega2560_UART0_115200_16000000L_B7_BIGBOOT.hex 262144 bytes):
```
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: auto set sck period (because given equals null)
avrdude: reading input file "C:\Users\furry-cheetah\Documents\Arduino\hardware\MegaCore-master\avr/bootloaders/optiboot_flash/bootloaders/atmega2560/16000000L/optiboot_flash_atmega2560_UART0_115200_16000000L_B7_BIGBOOT.hex"
avrdude: writing flash (262144 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 262144 bytes of flash written
avrdude: verifying flash memory against C:\Users\furry-cheetah\Documents\Arduino\hardware\MegaCore-master\avr/bootloaders/optiboot_flash/bootloaders/atmega2560/16000000L/optiboot_flash_atmega2560_UART0_115200_16000000L_B7_BIGBOOT.hex:
avrdude: load data flash data from input file C:\Users\furry-cheetah\Documents\Arduino\hardware\MegaCore-master\avr/bootloaders/optiboot_flash/bootloaders/atmega2560/16000000L/optiboot_flash_atmega2560_UART0_115200_16000000L_B7_BIGBOOT.hex:
avrdude: input file C:\Users\furry-cheetah\Documents\Arduino\hardware\MegaCore-master\avr/bootloaders/optiboot_flash/bootloaders/atmega2560/16000000L/optiboot_flash_atmega2560_UART0_115200_16000000L_B7_BIGBOOT.hex contains 262144 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 262144 bytes of flash verified
avrdude: reading input file "0x0f"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x0f:
avrdude: load data lock data from input file 0x0f:
avrdude: input file 0x0f contains 1 bytes
[... 46 lines cut ...]
 
Writing | ################################################## | 100% 34.25s

avrdude: 250812 bytes of flash written
avrdude: verifying flash memory against .pio\build\mega2560\firmware.hex:
avrdude: load data flash data from input file .pio\build\mega2560\firmware.hex:
avrdude: input file .pio\build\mega2560\firmware.hex contains 250812 bytes

avrdude: reading on-chip flash data:
 
Reading | ################################################## | 100% 25.37s
 
avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0x0c != 0x80
avrdude: verification error; content mismatch
 
avrdude: safemode: Fuses OK
 
avrdude done.  Thank you.
 
*** [upload] Error 1
========================= [FAILED] Took 317.68 seconds =========================
```
I think i do something wrong... Sorry, but i do it first time and many question ask in google :)

ellensp, 2020-04-20:
I've set up a test environment.
Firstly I can confirm the bug in the bootloader. 
I created two 248k intel hex files contain random data.
I uploaded the first file to a brand new mega. It uploaded as expected
I uploaded the second file to the same mega, and it gave me a  "verification error, first mismatch at byte 0x3c000"
I uploaded standard arduino mega2560 bootloader and then could upload once again. Just like you can.

I switched to opti bootloader and repeated the test.
Both test files uploaded without issues.

I added the following to platformin.ini in the  [env:mega2560]  section
board_upload.maximum_size = 261120 #smaller bootloader so can have larger programs. 
upload_protocol   = arduino 
upload_speed      = 115200

and it too uploaded as expected.

Can't replicate you last error...  

But..
i'm using a small firmware.bin

Can you upload your configs so i can generate a similarly large firmware?   

Also I am using vscode as atom is no longer official supported by platformio I would strongly advise you do the same.

furry-cheetah, 2020-04-20:
Thanks for a great job! 
My configs:
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/4503550/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/4503551/Configuration_adv.h.txt)

> Also I am using vscode as atom is no longer official supported by platformio.

I do this now and try to upload to board with optiboot again.
I think you a right, maybe i use a Atom with some bugs inside compiler.

> board_upload.maximum_size = 261120

Looks similar. I use new board in /boards/ folder with content:

```
{
  "build": {
    "core": "MegaCore",
    "extra_flags": "-DARDUINO_AVR_ATmega2560",
    "f_cpu": "16000000L",
    "mcu": "atmega2560",
    "variant": "100-pin-arduino-mega"
  },
  "frameworks": [
    "arduino"
  ],
  "name": "ATmega2560",
  "upload": {
    "maximum_ram_size": 8192,
    "maximum_size": 261120,
    "protocol": "arduino",
    "require_upload_port": true,
    "speed": 115200
  },
  "url": "https://www.microchip.com/wwwproducts/ATmega2560",
  "vendor": "Microchip"
}
```
One board on a printer now working, second i use for experiments.

ellensp, 2020-04-20:
Using your configs, sightly changed for latest bugfix 2.0.x, compiles and uploads via optiboot. No problem
```
avrdude: verifying ...
avrdude: 253926 bytes of flash verified 
```

furry-cheetah, 2020-04-20:
Super. I chenge Atom to VCstudio and try it now and update this post after it.

ellensp, 2020-04-20:
himm will have to examine your opti bootloader install logs 
In the mean time, here is a replacement stk500boot_v2_mega2560.hex to replace standard arduino one (freshly compiled and patched)   
[stk500boot_v2_mega2560.zip](https://github.com/MarlinFirmware/Marlin/files/4504714/stk500boot_v2_mega2560.zip)
GIve this a try (and remember to remove the extra opti boot lines from platformio)

furry-cheetah, 2020-04-20:
Ellensp, yes! You a super hero. 

Modified stk500 bootloader from you working pretty well. 
I 3 times flash firmware to test board and 2 times to real printer with eeprom clening - no any problems. You find a bug in arduino bootloader and working solution! 
And you bootloader looks little bit smaller 17307 bytes versus 21017 bytes arduino.

ellensp, 2020-04-20:
I first had to update the bootloader code to compile on modern avr-gcc compiler software. this produced smaller code.  (last update was 2012, so 8 years worth of compiler improvements!) 

https://github.com/arduino/Arduino-stk500v2-bootloader/pull/1 for details if your interested.  

But I think this is sorted, so if you could close the job, that would be good.

furry-cheetah, 2020-04-20:
Yep. My bug is catched and i not see any reason to stay it opened. Thank you very much again!

closed 2020-04-20 by furry-cheetah