MarlinFirmware/Marlin #17599: [FR] Support more than two serial ports

MCUdude, 2020-04-18:
### Description

Some boards, the Bigtreetech SKR 1.4 in my case has multiple serial ports. Currently, Marlin only supports two (SERIAL_PORT and SERIAL_PORT_2), but the hardware itself supports four. It would be great if there were possible to utilize at least one more serial port. In my case, I'm using USB (defined as serial port -1) for Octoprint, and hardware serial 3 for ESP3D (dedicated ESP8266/ESP01 header on the board). However, I'd also like to add a touchscreen to the printer, and this also requires a serial port. This means I'll have to sacrifice either Octoprint or my ESP3D.

borland1, 2020-04-19:
Besides the web UI, ESP3D supports serial/Wifi bridge using configurable port 8888.  So you should be able to use same port for Octoprint on pi.

charlespick, 2021-05-19:
3rd serial port has arrived

thisiskeithb, 2021-05-19:
Added in https://github.com/MarlinFirmware/Marlin/pull/21784. Closing.

closed 2021-05-19 by thisiskeithb