MarlinFirmware/Marlin #17600: Printer halts during filament change

axjurado, 2020-04-19:
Rumba board running marlin 2.0.5.3 in a custom build machine. In a 19hr print, at about 14 hrs into it, ned to change the filament. Printing from SD card, simplify3D sliced.

Pressed the change filament in the display menu and machine stopped. The LCD was not responsive. The print head stopped in the middle of the print and did not move to the park location.

After about a minute the LCD went back to the info screen, but it was not responsive. Steppers were disabled, but the machine would not respond.

Had to reset the machine and start over the print. 

Attached config file

[Marlin-2.0.x 14apr20.zip](https://github.com/MarlinFirmware/Marlin/files/4498269/Marlin-2.0.x.14apr20.zip)

Roxy-3D, 2020-04-19:
This was fixed a few days ago...   I think if you try it again with the latest BugFix it will work.
Please let us know if that is true.
I lost a long print this bug also.   (But it was only 8 hours in...   not 14 hours into the print.)
I know how frustrating it can be!!!!

axjurado, 2020-04-19:
Excellent... so just redo the config files with the bug fix ones?

Any way to automate transferring the values from one file to the other? 

Thanks!

Roxy-3D, 2020-04-19:
Well...    If your configuration files were up to date and current a week or two ago...   Just copy them on top of the current files.    But ExamDiff Pro is what I use to map my stuff to what ever is current.   There are also visual diff plug ins that work very well for Notepad++.  And Notepad++ is open source and free.   It really is pretty easy to use.

axjurado, 2020-04-19:
Apologize for the ignorance. I downloaded the bug fix config files and re-write the parameters for my machine. Is this the only thing I need to do and recompile/upload? Not sure what you meant by copying on top of the current files.

Roxy-3D, 2020-04-19:
Yes...   You just need to move the alterations in your configuration.h files over to the ones in the new source code tree.    In your case...   your files are probably current enough you can just copy the files over on top of the existing ones.

But Marlin is evolving very fast.   And configuration.h file options are added, deleted or renamed constantly.    It is always safer to do a visual diff of your files compared against what ever is in the current source trees.    You will see what is different and be able to very quickly decide if you need to understand something better.

axjurado, 2020-04-19:
Thanks. Hope last question. If I want to update the whole thing where do I get all the files? I don’t see where to download the bugfix whole set of files. Thanks again for your help

axjurado, 2020-04-19:
Never mind I found it in the marlinfw.org page. Thanks for your help

closed 2020-04-19 by axjurado