MarlinFirmware/Marlin #17607: [BUG] Printer not homing correctly after LCD/SD Stop Print

jreis1, 2020-04-19:
### Bug Description

ISSUE: When printing from SD card and request to Stop a print using LCD menu, only Y-axis is going home. X-axis stays where it is.   In the subsequent print, without resetting the machine, table  presents a huge offset, and assume homing almost in the centre of the table. 

### My Configurations
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4499565/configs.zip)

### Steps to Reproduce

1.  Start printing from SD card
2.  When printing, force stop print via LCD menu
3.  Start a new print

Expected behavior: after a stop via LCD,  "G28 XY" was expected.  
Actual behavior: as described above. 

#### Additional Information

marlin version: 2.0.5.3
Platform: Ender 3
mainboard: SKR v1.4 Turbo 
motor drivers: TMC2209

jreis1, 2020-04-23:
Guys, finally found the issue!!

I was setting the manual home position with a positive value, as: 
#define MANUAL_X_HOME_POS **+3.5**
#define MANUAL_Y_HOME_POS -7.5

This was affecting my stopping routine when pausing from the LCD, as  decribed above.  
As soon as I change that value to a negative number, stopping start to behave as expected.

thinkyhead, 2020-05-01:
How weird. It should be cool to set a home position as positive. However, this will make positions below your home (3.5) unreachable, so if any routines want to move to X0 they will grind.

Positive home positions are okay as a concept in Marlin, but they assume you have something like a Hall effect sensor that doesn't block the axis.

closed 2020-05-01 by thinkyhead