MarlinFirmware/Marlin #17612: [BUG] MOTHERBOARD BOARD_RAMPS_14_EFB

manotroll, 2020-04-19:
there is no option to heat the table in the slicer and on the LCD
and I also can't disable PIDTEMPBED for autopid

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4499988/Marlin.zip)

ellensp, 2020-04-20:
This is support question, not a marlin bug for feature request. 
Please read https://github.com/MarlinFirmware/Marlin/issues/14694

In saying that you have made an elementary error in your Configuration.h 
you have #define TEMP_SENSOR_BED 0
Which say there is no heated bed temperature sensor. Ie you have told the firmware you don't have a heated bed.

manotroll, 2020-04-25:
and now how does the table temperature offset?
is giving 10ºC difference with the thermometer

closed 2020-06-20 by boelle