MarlinFirmware/Marlin #17614: Upgrade to Marlin 2.x bricks Ender Silent Boards 1.1.5

lockheed, 2020-04-19:
I bought Ender 5 Pro with Silent Board 1.1.5. It had some old Creality Marlin on it (something like 1.1.6) so I upgraded it to 1.1.9. I already had bootloader installed, so it was an easy process.
I used the printer this way for couple of week and then decided to give Marlin 2.0.2 a try. So I flashed it just as the previous one. And that’s where problems started:

1.	8 out of 10 power ons, printer would not boot at all. The display turned on but was completely blank. Trying to get it on works sometimes, but it often takes so many power switch actions that I start worrying about mechanical/electrical damage it could cause. Naturally, resetting EPROM does nothing – neither from menu, nor gcode commands.
Other than – that the printer prints, but after every print once it homes the hotend, it crashes with EPROM ERROR on the otherwise blank display.

2.	You would think that something went wrong during flashing and I should be able to reflash the firmware. Logical. But no such luck. ArduinoIDE can no longer communicate with the printer! So what next?

3.	Bootloader reflash! So I re-flash the bootloader on the boards, flash Marlin again – any version I want -  and voila – it boots without any issues and no errors in sight… 
Until I discovered that all three axis appear to be completely dead. 

Running autohome from menu after few seconds produces:
```
Homing failed
PRINTER HALTED
Please reset
```

Moving axis individually from menu produces zero effect.

And homing from  gcode level goes like that:

```
>>> G28
SENDING:G28
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!
```

Some googling suggested printer might recognise endpoints as engaged and thus not moving. But after checking status printer returns they are not engaged.
4.	Contacted the seller who had no clue about that. We put it to a freak event. Eventually, he sent me a replacement Silent Board 1.1.5. So I flashed Marlin 1.1.9 on it and all was well for couple of months. Until I decided it’s time to upgrade to the newest Marlin 2.0.3…

5.	And now I am back to point 1. 

At least I can still use the printer in some capacity (when it decides to power on after ~10-20 powerons).

thinkyhead, 2020-04-20:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

lockheed, 2020-04-20:
I can't try any other version because upgrade to Marlin 2.x broke ArduinoIDE communication with the board.

sl1pkn07, 2020-04-20:
inlcude ISP (usbasp) pins?

ellensp, 2020-04-20:
Bricking the board is not very likely, The atmega1284p, like most 8 bit arduino boards, Has a section of the flash partitioned off as the bootloader. It is set in hardware that nothing can write to the bootloader area. Thus you need external hardware (ie a  avrisp) to over write it. It is much more likely that the controller has just died.  
But the board has a clearly labeled 6 pin ISP header, so the first thing is to try and put a new bootloader on it. Just to see if there is any life in the board at all. 
Also make sure to set your fuse bits correctly. Not doing this can for eg leave jtag enabled which breaks anything connected to IO pins that are stuck being a jtag.

lockheed, 2020-04-21:
I would appreciate if you guys read carefully the issue description. 
Flashing bootloader on this half-bricker board will render it completely useless, because although the software issues would go away, all axis motors will stop functioning at all, rendering half-bricked board to completely bricked.

And it does not look like a case of controller just dying, because it is already the second board with exact same issue after upgrading to Marlin.

thinkyhead, 2020-04-23:
What kind of error messages do you get back when using PlatformIO to do the upload? These might be a little more detailed. We also have the ability to massage many build settings through PlatformIO that we cannot through Arduino IDE, so I'd recommend you grab VSCode and the Auto Build Marlin extension and use these tools going forward. https://marlinfw.org/docs/basics/auto_build_marlin.html

lockheed, 2020-04-24:
Thanks for pointing me to PlatformIO. I tried compiling the source but what board for Silentboard 1.1.5 am I supposed to select when importing? I picked Sanguino 16 MHz, as it was the case when uploading with Arduino IDE, but build process fails with error:
https://pastebin.com/jbur1Uvd

sl1pkn07, 2020-04-24:
for the creality boards is melzi or meltzi_optimboot

lockheed, 2020-04-24:
When you were responding, I edited my post to update error messages with melzi board.

sl1pkn07, 2020-04-24:
yes, but is the same, needs update your config files

please share it

lockheed, 2020-04-24:
Yes, I introduced the config  change you mentioned and I am getting error from the last link. above.

That's the full config adv: https://pastebin.com/DGNFnHXv

sl1pkn07, 2020-04-24:
hwta version you use?

because now `HOMING_BUMP_MM` is

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/Configuration_adv.h#L611

and you use the old method

[image]

lockheed, 2020-04-24:
This one: 
[image]

Anyways, replaced the file with the one you linked and it compiled. Thanks!

So here's the error when trying to upload it to the printer - it's the same one as in ArduinoIDE:
https://pastebin.com/hCA4ekB2

sl1pkn07, 2020-04-24:
with usb or throuth the ISP port?

lockheed, 2020-04-24:
USB port. 
ISP port? Is it something that could be done with AdruinoUNO board?

sl1pkn07, 2020-04-24:
yep

lockheed, 2020-04-24:
Ok. Is there some howto on setting this up? My googling only finds Uno Bootloader flashing, not firmware.

ellensp, 2020-04-24:
read this. https://reprap.org/forum/read.php?158,871425

lockheed, 2020-04-24:
I followed it but now it can't compile:
https://pastebin.com/tt3KDhV2

lockheed, 2020-04-24:
Brilliant! After some trial and error I managed to flash the firmware this way and now the printer seems to behave as it should. Thanks a lot.

Two matters, though:
1. Do you think it is wise to now try the optiboot flash? I am hesitant as the last time I flashed a bootloader (regular one using ArduinoIDE once this above problem happened on my previous board) the axis stopped working completely.

2. Upgrading Marlin though ArduinoIDE software rather than PlatformIO is, I think, much more popular due to multiple howtos on youtube. And the problem there seems to be real as it happened to two of my boards. Perhaps it is worth looking into?

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle