MarlinFirmware/Marlin #17615: Skr v1.4 turbo not connecting to pc(pronterface,repetierhost,mattercontrol)

ahmetbozaci, 2020-04-19:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4500182/Configuration.zip)

ı have skrv1.4 turbo and marlin 2.0.5.3  Port is correct, baudrate is 115200 like in marlin but ı cant connect to pc with any program.
ı tried pronterface, mattercontrol, repetier host. nothing change.

ı can see sd card when ı connect board to pc with cable.ı can upload marlin without problem. but ı cant connected programs.

ı change baudrate on device manager to.
[image]

ellensp, 2020-04-20:
#define SERIAL_PORT 0   is wrong
you need #define SERIAL_PORT -1 to enable serial over usb
Also baud rates are ignored when using serial over usb.

ellensp, 2020-04-20:
Also This is support question, not a marlin bug or feature request. 
Please read https://github.com/MarlinFirmware/Marlin/issues/14694

closed 2020-04-20 by ahmetbozaci