MarlinFirmware/Marlin #17616: [FR] Allow retract after filament change

onovy, 2020-04-19:
### Description

M125 - Park nozzle currently doest this:
1. retract
2. park nozzle
3. wait for user
4. unpark nozzle
5. de-retract

During movement between printing position and parking position, there is no oozing, because filament is retracted.

M600 - Filament change currently does this:

1. retract
2. park nozzle
3. unload+load new filament
4. unpark nozzle

During movement from parking position to printing position, filament is oozing.

### Feature Workflow

I'm suggesting to add retraction just after loading new fiilament. When user confirms filament is pushed thru nozzle, user should clean nozzle. Then Marlin should retract (to lower pressure in nozzle), unpark nozzle back to printing position and un-retract (same as M125)

Currently it's really hard to snap hanging filament from nozzle "just in time" when nozzle is returning back to printing position.

Thanks.

MoellerDi, 2020-06-25:
@boelle @onovy it seems the related PR #17625 was merged. Did it resolve this FR?

onovy, 2020-06-25:
yes, thanks for pointing out

closed 2020-06-25 by onovy