MarlinFirmware/Marlin #17617: Support for GTM32 VD

dasch3a, 2020-04-19:
Hi,

I own a Geeetech A30 and realised that I've got the newer version with the GTM32 VD (not VB). So there is no support for this. Is there a solution in work and if not why? I think this would really help a lot of people. 

Thanks

David

ellensp, 2020-04-20:
Schematics can be found here https://www.geeetech.com/download.html?version_id=93 
[GTM32PROVDSCH.pdf](https://github.com/MarlinFirmware/Marlin/files/4503392/GTM32PROVDSCH.pdf)
From this a pins.h file can probably be created..

dasch3a, 2020-04-20:
I would if I could. But unfortunately I'm not really good in programming. And I have no clue of these schematics. So when there is a schematic of this board too why is there no support from Marlin? Or this there something in Progress? (hopefully)
Greetings

chepo92, 2020-04-28:
> 
> 
> I would if I could. But unfortunately I'm not really good in programming. And I have no clue of these schematics. So when there is a schematic of this board too why is there no support from Marlin? Or is there something in progress? (hopefully)
> Greetings

Marlin is maintained by community and you may be the first with that board, but you may help, first checking the differences between REV D and B schematic as it may not be many differences, with the differences known, if any, we can make a new pin.h

sjasonsmith, 2020-04-28:
@Vertabreak works with a lot of Geeetech stuff. I haven't seen him as much lately, but he might know something. You may also go look at the Geeetech forum to see if anyone there knows of anything in progress for this board.

chepo92, 2020-04-30:
Looking at the differences, most of the vital parts are quite the same, extruders, hot ends, limit switches, they added some conectors to add sensors, I don't have the RevH so can't test/check

Main changes: 
POWER: 
- LCD power was not controlled but with a Lm358. Now SPI1_NSS drives the FET (that may be needed if LCD is used). Also note that the drived line was 12V, now 5V
- Removed MC7812BD

STEPPER
This is one may be a vital change, may be a bug or something, in stepper driver E1_A1, STEP pin has label SPI1_SCK, and has a E1_STEP label to the left, E1_STEP is used in BL Touch. 
- Different header connection

USB to serial 
- now is a shady cheap CH340C

POWER DETECT (new)
- This is new, resistor divider network wired to X_MAX

EXTERUDER (no change in name :D)
- Added filament detect connectors, wired to SPI1_MISO, SPI1_MOSI and Y_MAX

UART LCD to FPC_CON
- Changed 10 pin to 12 pin header

WIFI (removed)
- Removed the circuit

3DTOUCH (added)
- added two connectors, E1_STEP should be driving the "servo" of 3dtouch, and Z_MIN wired to sensor signal. 

This may be implemented as a board variant and condition if it is a revB or revH to swap the pins.
I don't have a revH so can't check those changes, board donations are welcome :D

Vertabreak, 2021-02-02:
I have the board and the current vb pins work on it just fine. should probably add a note for - this to the pins file in one of my pull requests. But on that note gtm32 support is not exactly end user viable with a number of issues making printing very sketchy.

Vertabreak, 2021-04-02:
this can be closed merged pins for VD.

closed 2021-04-02 by thisiskeithb