MarlinFirmware/Marlin #17618: Tool for understanding: feedrate, acceleration and jerk

daleckystepan, 2020-04-19:
Hi guys,

I've written basic app to better understand basic motion principles in Marlin. It is very tough for begginers.

https://github.com/daleckystepan/MarlinMotion

https://github.com/daleckystepan/MarlinMotion/raw/master/screenshot.png

It is written in Python, based on simply RungeKutta ODE solver. Jerk is used only for initial velocity which is half of jerk. It is only for one axis now. We can improve it by proper jerk, s curve acceleration or so on.

Feel free to test it and give me feedback.
Thank you

tomarigr, 2020-04-20:
Yeah but why jerk and not junction deviation?

daleckystepan, 2020-04-20:
Because I don't know how jerk and JD work exactly. If there is someone who is able to explain this, just do it.

closed 2020-06-20 by boelle