MarlinFirmware/Marlin #17626: [BUG] G76 failed message.

TB1405, 2020-04-20:
### Bug Description
activated the temp compensation in the config files and started G76. 
everything works fine the extruder goes to the defined positions and the temp of the bed increases by 5° step by step . 
But by max bed temp the error message appears and the printer restarts. (if i set the max temp to 100°C the same error comes at 100C°)

The last lines from octoprint terminal 
`Recv:  T:34.87 /0.00 B:90.01 /90.00 P:39.91 /0.00 @:0 B@:65 T:34.83 /0.00 B:89.99 /90.00 P:39.98 /0.00 @:0 B@:68
Recv: echo:busy: processing
Recv: Error:Probing Failed
Changing monitoring state from "Operational" to "Error: Probing Failed"`

### My Configurations

[Configuration_bugfix.zip]

### Steps to Reproduce

1. [Download the bugfix from today and my config files]
2. [load to printer]
3. [run G76]

#### Additional Information

one more question. 
https://marlinfw.org/docs/features/probe_temp_compensation.html#saving-and-loading 
How could i edit the values in the terminal ? 
maybe somebody could make a example. 
I want to use the example values from the page for my printer.

boelle, 2020-06-20:
@TB1405 is the issue still there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle