MarlinFirmware/Marlin #17627: M502 M500 2X Req to save new firmware into EEPROM on update, SKR PRO

DroneMang, 2020-04-20:
### Bug Description

Hi,

I noticed this while setting up sensorless homing.  If I create a new firmware and change the default values in the firmware and then compile, I must run M502, M500, M501 twice to see the changes.  If there is an EEPROM mismatch, this also persists until the second set of M502, M500, M501.  

In the SKR PRO pins file, does this have to do with the default FLASH based EEPROM?  Could there be more of a description in the pins file as to how each of these options work, FLASH based and SRAM based?

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. run M502, M500, M501 twice to see the changes

**Expected behavior:** run M502, M500, M501 once.

**Actual behavior:** [What actually happens]

BastR, 2020-04-22:
I have the same problem but this one has been around for months...It affects a lot of SKR PRO users, I know 3 other people who use this card and we have exactly the same problem.

Moreover and randomly, the BLT doesn't work anymore after an M502 M500 as seen in this bug report https://github.com/MarlinFirmware/Marlin/issues/16169

sjasonsmith, 2020-05-09:
I saw the same issue yesterday, with two M500 calls being required to write to the flash.

I did NOT see the issue if I added `#define FLASH_EEPROM_LEVELING`.

Debugging the code it appears that the command to erase the flash sector is failing, which is preventing the write from occurring. It then works on the second try. I don't know the reason for the failure the first time, I haven't debugged that far.

`FLASH_EEPROM_LEVELING` hides the problem, because it only erases the sector once in about 30 saves. It _might_ still have the same issue when it does the erase, but it did not occur the one time I force it to wrap around and erase.

sjasonsmith, 2020-05-10:
I was about to post a pull request to fix the double-M500 issue, when I thought to test the BLTouch issue @BastR mentioned. Much to my surprise, he is correct! After writing to the flash my BLTouch stops responding.

I verified with a logic analyzer that the servo output goes to a 100% duty cycle after writing to the flash.

DroneMang, 2020-05-16:
This has now been resolved in the newest bugfix.  So I am closing this.  Thanks.

closed 2020-05-16 by DroneMang