MarlinFirmware/Marlin #17629: STATUS_BED_X' was not declared in this scope

Enric580, 2020-04-20:
I am trying to install the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER screen on an MKS_SGEN_L board
when i compile marlin 2.0.x I get this error

Marlin \ src \ lcd \ dogm \ status_screen_DOGM.cpp: 253: 30: error: 'STATUS_BED_X' was not declared in this scope; did you mean 'STATUS_FAN_X'?
   253 | const uint8_t bx = STATUS_BED_X + STATUS_BED_WIDTH;

swilkens, 2020-04-20:
Please try again with the bugfix branch and let us know if your issue still exists:

Firmware:
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

Configuration:
https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x

Enric580, 2020-04-21:
OK, the problem has been fixed, it works fine.
Thanks

ellensp, 2020-04-21:
Can you close the job then? Thanks.

closed 2020-04-21 by Enric580