MarlinFirmware/Marlin #17630: [BUG] HYBRID_THRESHOLD M913 X200 -> Stores as 201

swilkens, 2020-04-20:
### Bug Description

Using `HYBRID_THRESHOLD` and setting the limit value via `M913` seems to store it as a different value for some input ranges.

Configure as X150 -> stores as X152
Configure as X200 -> stores as X201 

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4505535/Marlin.zip)

### Steps To Reproduce

1. Have `HYBRID_THRESHOLD` enabled in `Configuration_adv.h`
2. Do `M502` to load defaut values
3. Notice in `M503` output:
>echo:; Hybrid Threshold:
>echo:  M913 X100 Y100 Z3
4. Do `M913 X150`
5. Notice in `M503` output:
> echo:; Hybrid Threshold:
> echo:  M913 **X152** Y100 Z3
6. Do `M913 X200`
7. Notice in `M503` output:
> echo:; Hybrid Threshold:
> echo:  M913 **X201** Y100 Z3

**Expected behavior:**
I would expect the inputted value to be stored.

**Actual behavior:**
The value is modified, seemingly arbitrarily. 

#### Additional Information

Bugfix 2.0.x (https://github.com/MarlinFirmware/Marlin/commit/abd42e5dea4670fd3f4c9177caabb31f391edbeb)

swilkens, 2020-04-20:
Additionally, if I do:

`M913 X200 Y200 Z200`
`M913 T0 E200`

I get:

`M913 X201 Y201 Z219`
`M913 T0 E202`

So it seem to behave differently per axis.

All axes have the same stepper driver (TMC2209), they only differ in terms of steps/mm. X/Y have the same steps/mm (and show the same deviation), Z and E have different steps/mm and also show different deviations.

Is this by design?

thisiskeithb, 2020-04-21:
I see this on my builds as well. I think it came up in another issue a while back and had to do with accuracy getting lost in how those values are stored & displayed.

teemuatlut, 2020-04-26:
Not a bug.
If you divide 7 with 2, you get 3.5 which is stored as an integer 3.
Then to get back you multiply 3 with 2 and you get 6.

swilkens, 2020-04-26:
@teemuatlut Ok, I understand that float -> int loses a decimal, but then this means this configuration parameter is not stored as mm/s (what we put in), rather as something else? (steps / planner cycle perhaps?)

For a user point of view, this becomes confusing; what value should I now put in my slicer to be above or under my `HYBRID_THRESHOLD`? The value I put in manually (because that will go through the same match chain -> same deviation?) or the value that's echo'd back to me after saving? Apparently this also depends on steps/mm for the axis, so I would have to check per axis?

swilkens, 2020-05-11:
Related:

https://github.com/MarlinFirmware/Marlin/issues/17080

boelle, 2020-06-20:
@swilkens is this still an issue?

swilkens, 2020-06-21:
I would say this is still an issue. I understand that this is caused by an internal conversion, but it's still confusing to the end user to see a configuration value be modified before storing.

If I set 200 on Z, it's stored as 219. What is now my actual threshold? If I do testing on my printer to find out the threshold value at which the pritner starts skipping steps, how should I then set the threshold?

Say I find 200 is my threshold on Z. I want the printer to switch to spread at 200. But setting it at 200 saves it as 220? Am I now still risking missing steps?

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle