MarlinFirmware/Marlin #17631: [BUG] error: 'X_HOME_BUMP_MM' was not declared in this scope

willis7, 2020-04-20:
### Bug Description

Failed build when using "Auto Marlin Build" and "PlatformIO" in VSCode.

### My Configurations

commit `0518dec60`
https://github.com/MarlinFirmware/Marlin

commit `7db6050`
https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-5

### Steps to Reproduce

1. Clone marlin project
2. Copy and paste; `Configuration.h`, `Configuration_adv.h`, `_Bootscreen.h` and `_Statusscreen.h` at commit `7db6050` for Creality Ender 5.
3. Build `sanguino1284p` in "Auto Marlin Build".

**Expected behavior:** [What you expect to happen]
Successful compile and link.

**Actual behavior:** [What actually happens]
Compiling fails with the message below.
```
...
Compiling .pio/build/sanguino1284p/src/src/module/printcounter.cpp.o
Compiling .pio/build/sanguino1284p/src/src/module/probe.cpp.o
Compiling .pio/build/sanguino1284p/src/src/module/scara.cpp.o
Compiling .pio/build/sanguino1284p/src/src/module/servo.cpp.o
Marlin/src/module/motion.cpp:77:71: error: 'X_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                       ^
Marlin/src/module/motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^
Marlin/src/module/motion.cpp:77:80: error: 'Y_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                ^
Marlin/src/module/motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^
Marlin/src/module/motion.cpp:77:89: error: 'Z_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                         ^
Marlin/src/module/motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^
Compiling .pio/build/sanguino1284p/src/src/module/stepper.cpp.o
*** [.pio/build/sanguino1284p/src/src/module/motion.cpp.o] Error 1
Compiling .pio/build/sanguino1284p/src/src/module/stepper/L64xx.cpp.o
=================================================== [FAILED] Took 50.46 seconds ===================================================

Environment               Status    Duration
------------------------  --------  ------------
mega2560                  IGNORED
mega1280                  IGNORED
rambo                     IGNORED
FYSETC_F6_13              IGNORED
...
```

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

swilkens, 2020-04-20:
Try with the configs from the `bugfix-2.0.x` branch:

https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x

willis7, 2020-04-20:
Thank you, that has indeed fixed the issue. I will experiment with the bugfix branch for now, if I find any issues I shall revert and provide feedback.

closed 2020-04-20 by willis7

tablatronix, 2020-04-22:
I still get this for ender-3 config, both repos are on bugfix-2.0.x branches.

```PHP
Compiling .pio/build/sanguino_atmega1284p/src/src/module/printcounter.cpp.o
Marlin/src/module/motion.cpp:73:71: error: 'X_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                       ^
Marlin/src/module/motion.cpp:79:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^
Marlin/src/module/motion.cpp:73:80: error: 'Y_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                ^
Marlin/src/module/motion.cpp:79:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^
Marlin/src/module/motion.cpp:73:89: error: 'Z_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                         ^
Marlin/src/module/motion.cpp:79:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^
*** [.pio/build/sanguino_atmega1284p/src/src/module/motion.cpp.o] Error 1
```

tophat17, 2020-06-14:
I was able to get 1.1.7 running on my Anet a8. ANY version most recent then that would cause issues.

thinkyhead, 2020-06-15:
@tophat17 — If you're looking for a conversation about your 3D printing experience, there are better forums for that:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y