MarlinFirmware/Marlin #17632: [BUG] g28 homing not using safe home position with bltouch probe installed

shrpshtr325, 2020-04-21:
Added BLtouch auto bed leveling probe, and when i enabled the BLtouch (including safe home) the homing stopped functioning as excepted.  When i use a g28 the printer homes x and y fine, then trys to home z off the build platform (at the x and y endstops), if i do g28 x y and the g28 z same thing happens.  However, if i use g28xy the move the printer head onto the bed (g1 x30 y20) then g28 z the printhead moves to the safe home position and homes as expected. 

my current workaround to print is to append the following to the beginning of every gcode file (in place of a simple g28 and g29)

G28 X Y
G1 X100 Y100
G28 Z
G1 X0 Y0 Z0
G29

[Marlin_ts.zip](https://github.com/MarlinFirmware/Marlin/files/4507067/Marlin_ts.zip)

ktims, 2020-04-25:
I am having a similar issue. It seems that Marlin *thinks* it has moved to the safe homing position (it reports x,y as being bed centre, and moves toward the endstops crash the machine), but it never actually moves away from x=0,y=0.

I am running 2.0.5.3, and maybe it is a contributing factor but I am not using ZMIN for the BLtouch, as my machine has two Z motors and two Z endstops, so I have kept them both.

shrpshtr325, 2020-04-26:
Now that ypu mention it I went and checked, this is exactly what is happening to me too

ktims, 2020-04-27:
I think I might have sorted this out in my case. It appears that having `USE_XMAX_PLUG` defined actually means something more like `USE_XMAX_ENDSTOP`. I had set this since XMAX is where I have plugged my touch probe (as the comment suggests). However this seems to have messed up the homing procedure.

With this undefined, the machine does safe homing properly, and the machine origin doesn't get messed up like it did before.

shrpshtr325, 2020-04-27:
I will try the bugfix  branch when i get my replacement board in.  I had a bad crimp on the power plug of my ender3 and didnt realize it, used the factory power plug and it appears to have cooked my new board, replacement due in Wed or Thurs.  

i do not have any of the max_endstop lines enabled in my config files.

adminmat, 2020-04-28:
I've been experiencing this as well. Flashed 2.0.5.3 and even though I have Z_SAFE_HOMING set to 60 for both X and Y a G28 first moves X and Y to endstops followed buy a brief Z movement before showing "Z Probe Past Bed". I do not have USE_XMAX_PLUG defined. 

Should I understand that using a bugfix 2.0.x will fix this?

shrpshtr325, 2020-05-16:
> Can you try with the latest bugfix-2.0.x branch? I updated one printer to it a few days ago and my safe homing is working fine.

finally got the bugfix branch to compile (i had to completely redo the configuration files, the ones i made from 2.0 wouldnt compile and the errors were seemingly meaningless), and the safe homing now works as expected.

sjasonsmith, 2020-05-16:
@shrpshtr325 usually messages are added to the errors when configuration options change, to help guide people toward the needed changes.

Can you provide examples of the changes you had to make, which did not provide clear error messages?

shrpshtr325, 2020-05-16:
the only error i recall getting was acr-gcc folder not found, i could not get rid of the error, i had to take the new configuration files and re-edit them to match my setup, unfortunately i didnt take any notes, but the solution was to throw out the older configuration files and modify the new ones included in the bugfix download (which supposedly isnt necessary)

closed 2020-05-16 by shrpshtr325