MarlinFirmware/Marlin #17633: HOT END 1 MAX TEMP Fault when upgrading from 2.0.1 to 2.0.5.3 / Bugfix versions

RFBomb, 2020-04-21:
### Bug Description
I was running v2.0.1 and went to upgrade to 2.0.5.3 // the bugfix version and found that after the upgrade, my printer all of a sudden was reporting a  "MAX TEMP E1" error immediately on bootup. 

Thinking it was my sensor on the way out (I had damaged the wires changing the hot end), I swapped it out for a direct replacement. Still got the fault. 

### My Configurations

SKR 1.3 on Ender 3.

default_envs = LPC1768 in Platformio.ini

New configs and 2.0.1 configs are uploaded. 
[Config_2.0.1.zip](https://github.com/MarlinFirmware/Marlin/files/4507503/Config_2.0.1.zip)
[Config_2.0.5.3.zip](https://github.com/MarlinFirmware/Marlin/files/4507505/Config_2.0.5.3.zip)

### Steps to Reproduce

All I did was upgrade my firmware. 

**Expected behavior:** [What you expect to happen]

Smooth upgrade to new firmware, no faults that arent real.

**Actual behavior:** [What actually happens]

MAX TEMP fault

#### Additional Information

Reverting to 2.0.1 clears the fault. 
As far as I can tell, nothing changed regarding pin assignment for this board (though structure of that file has changed).

I'm at a complete loss as to the issue here, as I cannot find out  what is wrong.

RFBomb, 2020-04-21:
It seems to be a nearly identical issue to this, and I tried everything he did, but still have no clue why it does this on newer firmware. 

https://github.com/MarlinFirmware/Marlin/issues/17286

thisiskeithb, 2020-04-21:
> I had damaged the wires changing the hot end

You likely shorted the thermistor which would cause the reading to be max. Time for a new board.

RFBomb, 2020-04-21:
You must have missed the part where moving to old firmware doesn’t have the
issue and it reads temperature fine.

RFBomb, 2020-04-22:
OK, so after 2 days of troubleshooting the issue, replacing the thermistor, many times recompiling several different versions with a bunch of config edits, all of a sudden its working fine on 2.0.5.3, despite it erroring out on a previous build of identical version # using that configuration. 

Since @thisiskeithb  recomended I grab a new board, I decided to test out my board's temperature pins, since SKR has 2 inputs and I currently only need 1. 
I added this to my configuration.h for the test run (but am leaving it commented out so I still use the default pins once I get it back together. But this allowed me to verify both hot end sensors work properly and read the same values, which they do. New board not required. 
```
/* Temperature Sensor DEBUG TESTING 
 * Swap Temperature Pins by uncommenting these lines. (If commented, then it uses standard pin definition)
 * Temp_Sensor_0 will then read from Sensor_2 input, and vice versa. 
 * Pin notes taken from pins_BTT_SKR_common.h
 */
//#define TEMP_0_PIN                    P0_25_A2  // A1 (T1) - (68) - TEMP_0_PIN
//#define TEMP_1_PIN                    P0_24_A1  // A2 (T2) - (69) - TEMP_1_PIN
```

The 'broken thermistor' I mention in the beginning of my issue was unbroken prior to troubleshooting the issue, and the alarm even after I cut the thermistor wires was only present on firmware v2.0.5.x, not on v2.0.1.  (Cut them since I planned to replace it with a better one anyway). 

SO long story  short: I have no idea wtf caused this headache, but for now it is working?

closed 2020-04-22 by RFBomb

mdellen, 2020-05-27:
Exact same problem here. HOT END 1 MAX TEMP Fault when upgrading from 2.0.1 to 2.0.5.3 / Bugfix version. When moving back to 2.0.1 everything is fine. Deleted the .pio directory, run cleanup and update all, recompiled and issue persists. Might be related to: https://github.com/MarlinFirmware/Marlin/issues/14139 
https://github.com/MarlinFirmware/Marlin/issues/15834
https://github.com/MarlinFirmware/Marlin/pull/14798

No solution at the moment, will see if BOGUS_TEMPERATURE_FAILSAFE_OVERRIDE works to confirm it is the same issue.

RFBomb, 2020-05-27:
Hey,

FWIW, I fixed this issue and my ender 3 w/ skr1.3 is currently running
2.0.5.3 fine. I did end up putting a separate new thermistor in it, but
I’ll have to double check my order date to see if that was before I
reported it working again.

Changing the hot end thermistor type from a 1 to a 5 may be what fixed it
for me, now that I think about it. But admittedly, I cannot recall if I put
the new thermistor in and changed the value to 5 then it was fixed, or if
it was fixed simply with a new thermistor. I believe it was changing the
type that fixed it.

Please change your type from 1 to 5 (assuming you are using type 1) and
test it out to see if the problem persists. (I wouldn’t leave it there if
it doesn’t match your setup, but it’s worth it to see if that’s the issue
at hand).

mdellen, 2020-05-29:
> Hey, FWIW, I fixed this issue and my ender 3 w/ skr1.3 is currently running 2.0.5.3 fine. I did end up putting a separate new thermistor in it, but I’ll have to double check my order date to see if [...]

Thnx, tried you solution but error remains. 

When compiling the exact same source code (BTT 2.0.1 SKR 1.4 turbo branch) today I get the E1 max temp error. When I load the .bin from a few months ago (same source) everything is fine. So it seems something has changed in the libraries and not Marlin itself.

I need to change something in the config but am not able to compile new firmware without getting the error. Tried BTT 2.0.1 branch, 2.0.1, 2.0.4.4 and 2.0.5.2

I will have to dig deeper, think maybe something in the ADC part has changed?

RFBomb, 2020-05-29:
That’s strange. I was able to Re-compile the official marlin v2.0.1 and use
it just fine when I was troubleshooting.

The original bin 2.0.1 bin worked, a recompiled marlin 2.0.1 worked, but
2.0.5.3 gave the error.

I feel like uncommented  something in 2.0.5.3 to fix it. I’ll upload my
config.h files for you to compare against yours and see what major
differences are. (Notepad ++ has a great compare plugin)