MarlinFirmware/Marlin #17635: [BUG] Virtual Tool 1 is Incorrectly initialized with 0 : 0 mix on 2 in 1 mixing printer

lterrill, 2020-04-21:
### Bug Description

Beginning with Marlin 2.0.5, Virtual Tool 1 is incorrectly initialized with 0 : 0 mix. Attempting to extrude using tool 1 with this default mix results in printer crash. 

### My Configurations

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4510264/Configs.zip)

### Steps to Reproduce

Using a Geeetech A10M with 2 in to 1 mixing hot-end 
1: Boot printer
2: Examine Virtual Tool mix via LCD. Note that Virtual tool 1 (second tool) is initialized with a mix of 0 : 0. 
3: G28; home
    T1; select tool 1
    G1 E5 F50; extrude using tool 1
4: printer locks up, crash

**Expected behavior:** [What you expect to happen]

Virtual tool 1 previously initialized with a mix of 0 : 100. This resulted in the extrusion move using the second physical filament extruder

**Actual behavior:** [What actually happens]

Beginning with Marlin 2.0.5, Virtual tool 1 initializes with a mix of 0 : 0. I believe this results in some bad math that crashes Marlin

#### Additional Information
[image]

Patag, 2020-04-21:
Don't know exactly your hotend configuration, but I guess you should use SINGLE_NOZZLE (2 steppers, 1 heater and 1 nozzle) rather than MIXING_EXTRUDER.

thinkyhead, 2020-04-21:
I saw the report on FB. I'll patch it asap.

closed 2020-04-21 by thinkyhead

iatwalrus, 2020-04-25:
Has a patch or fix been put in for this? If so, where may I find it?

lterrill, 2020-04-26:
> Has a patch or fix been put in for this? If so, where may I find it?

bugfix-2.0.x branch contains the patch that corrects this issue in mixing.cpp, reset_vtools(). I expect this patch will be merged into the 2.0.x branch with the next release on that branch, but that had not yet occurred as of 4/26/2020

commit 95a3b868 (2020-04-21): Fix Mixer::reset_vtools bug

Fixes #17635
commit a5b2e3e6 (2020-04-21): Fix Mixer::reset_vtools bug

Fixes #17635