MarlinFirmware/Marlin #17636: [BUG] Cannot quit Invaders game Marlin 2.0.5.3 - Ender 3 - SKR mini E3 v1.2

Tygrys-1, 2020-04-21:
### Bug Description

After entering Invaders game there is no way to quit it other then switching the power off. According to the manual long press of the dial should quit the game. This does not work.

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4510592/Marlin.zip)

### Steps to Reproduce

1. Enable games in configuration
2. Start Invaders game
3. Try to quit

**Expected behavior:** [What you expect to happen]

Long pressing the dial should quit the game

**Actual behavior:** [What actually happens]

Long pressing the dial constantly fires bullets.

boelle, 2020-06-20:
@Tygrys-1 is the issue still there?

Tygrys-1, 2020-06-21:
If you ask for 2.0.5.3 it still is there. I have not tried the bugfix version, it requires to much changes in the configuration for me. I'll wait for the next release and let you know. I do not see any PR mentioning this issue.

boelle, 2020-06-21:
always test with the latest

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

thisiskeithb, 2024-08-30:
This bug still exists.