MarlinFirmware/Marlin #17637: [BUG] Strange Neopixel collors during the heating phase Marlin 2.0.5.3 - Ender 3 - SKR mini E3 v1.2

Tygrys-1, 2020-04-21:
### Bug Description

I have 4 WS2812 leds connected to the Neopixel output on the board. Startup animations, white light during the print and green light after print is working ok. On starting the bed heatup phase the leds turn mostly blue, but there are random green or cyan or even black leds. This artifacts are changing over time. During the heatup phase leds are turning into violet, but some stay blue or change to other colors from time to time. It is not as sugested in manual "sequential changing" when more leds will go violet. The same situation is when heating the bed - the colors are mostly violet and red, but with artifacts of other collors as well.

### My Configurations
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4510681/Marlin.zip)

### Steps to Reproduce

1. I am using this heatup sequence to speed things up:
M140 S60 ; start preheating the bed
M104 S200 ﻿T0 ; start preheating hotend
G28 ; Home all axes
M190 S60 ; heat to Cura Bed setting
M109 S200 ﻿T0 ; heat to Cura Hotend

2. The leds will go different collors in random order, trying to be from blue to violet and then red, but at non-consistant way and with color artifacts.

**Expected behavior:** [What you expect to happen]

Leds should start blue and change one by one to violet during heating the bed, and then change from violet to red while heating the hotend.

**Actual behavior:** [What actually happens]

Described above.

#### Additional Information

Seems like the timing issue on driving the WS 2812. On heatup the main task is to check the temperatures and this may affect the led driver timing.

vivian-ng, 2020-04-23:
I don't have a SKR Mini E3 so I cannot help, but it is basically a library support issue. The Adafruit_Neopixel library used by Marlin needs to support the microcontroller you are using. For the SKR Mini E3, which I think uses a STM32F103 MCU, support doesn't seem to be there yet.
[Adafruit_NeoPixel Supported Chipsets](https://github.com/adafruit/Adafruit_NeoPixel#supported-chipsets)

On a related note, it seems that BIGTREETECH may have its own version of the library that you can try.
[Feature request Neopixel STM32F4 support for Neopixel Library](https://github.com/adafruit/Adafruit_NeoPixel/issues/198#issuecomment-526834220)

Tygrys-1, 2020-04-23:
I am using the BTT library. Seems like the library issue. The LED timing is done with asm(nop), so it will be no accurate in case of interrupts or maybe something like DMA or whatever. The current Neopixel implementation from Adafruit seems to support STMF32, but importing it to Marlin rises compilation errors with lacking definition of GPIO_TypeDef in Adafruit library code. Otherwise it seem to be simmilar to the BTT version. Can this definition be added somehow?

boelle, 2020-06-20:
@Tygrys-1 is the issue still there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle