MarlinFirmware/Marlin #17638: [BUG] E0_AUTO_FAN_PIN and CONTROLLER_FAN_PIN do not work

android444, 2020-04-21:
I have downloaded today's version of MARLIN BUGFIX-2.0.x.
The only thing I changed in the CONFIGURATIONS.H file is
#define MOTHERBOARD BOARD_SILVER_GATE
#define SILVER_GATE_GLCD_CONTROLLER

I ruled out the SILVER_GATE board failure.
I changed the default pins number for
E0_AUTO_FAN_PIN
CONTROLLER_FAN_PIN
on
FAN_PIN

Part of the pins_SILVER_GATE.h file
#ifndef FAN_PIN
  #define FAN_PIN                              2//3//5
#endif

#define HEATER_0_PIN                           7

//#ifndef E0_AUTO_FAN_PIN
 // #define E0_AUTO_FAN_PIN                      3
//#endif

//#define CONTROLLER_FAN_PIN                     2

The fans should turn on and not work.

ellensp, 2020-04-22:
From what you have provided you have no E0_AUTO_FAN_PIN or  CONTROLLER_FAN_PIN defined. You have commented them both out.

android444, 2020-04-22:
Here I checked if the hardware is ok. Both are defined.

ellensp, 2020-04-22:
For anyone looking this is a board he has designed himself https://github.com/android444/Silvergate
Using a, never before tested with marlin, atmega2561

android444, 2020-04-22:
What does this change?

ellensp, 2020-04-22:
many many things....
First thing I notice when trying to compile for a atmega2561 is platformio pulls in framework-arduino-avr-megacore 
I suspect its using the megacore pin definitions.. https://camo.githubusercontent.com/2e9fe11875f8ca873e6d9ed5c9602acdaf5c089f/68747470733a2f2f692e696d6775722e636f6d2f737765524a73332e6a7067 
So D2 is PE2 and D3 is PE3 instead of D2 being PE4 and D3 being PE5 as you have wired it. 
So check those pins for change..  or just set to the fans to D4 and D5 and see if it works.
This is only a guess...  funnily enough I don't have any 2561 boards to try it on.

android444, 2020-04-22:
It's not what you write.
Familiarize yourself with it
https://github.com/android444/Silvergate/tree/master/Arduino%20silvergate
and pinology in Marlin
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/HAL/AVR/fastio/fastio_1281.h

android444, 2020-04-22:
It is a bit strange because a week ago when I compiled it CONTROLLER_FAN_PIN works fine.
Someone makes changes but doesn't check if it works.

boelle, 2020-06-20:
@android444 is the issue still there?

android444, 2020-06-21:
Nobody accepted.

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle