MarlinFirmware/Marlin #17641: [BUG] Software Endstops Only Work After Switching Modes

Chronicmemecrafter, 2020-04-21:
### Bug Description
Okay, so I believe I found a really weird bug. I'm using an SKR Mini E3 V1.2 with a TFT35 E3 V3.0. I have my software set so that the LCD boots to TFT mode. In my configs I have software endstops enabled as well. But when I go to manually move my axis's (whether or not it was homed) it acts as though software endstops aren't enabled, as in the numbers go into the negatives. (To combat this, for now, I have turned on 'endstops trigger even when not homing' so that I don't frick up the motors. I have tried adding a custom M211 S1 command into the firmware for the TFT to no avail, it says they're on but nothing changes. The weird part is if I switch to marlin mode and try to move the axis into the negatives it works as it should. And if I go back after that it works in the TFT too!

### My Configurations
Custom Marlin 2.0.5.3 build config: https://pastebin.com/qFFDd4hi
Custom Marlin 2.0.5.3 build config_adv: https://pastebin.com/SPYYANsw
Custom TFT35 E3 V3.0.26 build config: https://pastebin.com/JvDgVPfk
### Steps to Reproduce
1. Use same hardware and possibly same firmware
2. Set TFT to boot to TSC mode
3. Try to move into negatives manually
4. Switch to marlin mode 
5. Try to move into negatives
6. Go back to TSC mode
7. Try to move to negatives

**Expected behavior:** [What you expect to happen]
Since software endstops are enabled, it should just stay at zero when I try to go back further right after boot on TSC mode

**Actual behavior:** [What actually happens]
Goes into negatives and slams hotend into side unless you have 'endstop still active when not homing enabled'
Works after you go into Marlin mode and go back

#### Additional Information

None unless any additional questions.

boelle, 2020-06-20:
@Chronicmemecrafter is the issue still there?

Chronicmemecrafter, 2020-06-20:
@boelle no sorry I’ll close it

closed 2020-06-20 by Chronicmemecrafter