MarlinFirmware/Marlin #17643: Can not compile vs 2.0.5.3

wr-rider, 2020-04-21:
I Took the clean vs and was able to compile it. I only changed the main board to BOARD_BTT_SKR_V1_4_Turbo and set the pio to lpc1769 and no mater how many times I tried it would not compile and eventually would crash my Win10 machine. The best I could see before it crashed was it couldn't connect to some server. Sorry I can't be of more help. My older vs 2.0.5.2 still works.

# NO SUPPORT REQUESTS PLEASE

Support Requests posted here will be automatically closed!

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. See https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question.

For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org Marlin Forum http://forums.reprap.org/list.php?415
- Tom's 3D Forums https://discuss.toms3d.org/
- Facebook Group "Marlin Firmware" https://www.facebook.com/groups/1049718498464482/
- Facebook Group "Marlin Firmware for 3D Printers" https://www.facebook.com/groups/3Dtechtalk/
- Marlin Configuration https://www.youtube.com/results?search_query=marlin+configuration on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a Bug Report at https://github.com/MarlinFirmware/Marlin/issues/new/choose).

thisiskeithb, 2020-04-21:
Try cleaning your install (click the trash can icon) and build again. GitHub was timing out earlier today, which would cause the “not found” errors.

ellensp, 2020-04-22:
no error messages, no configs, no way to help you. It does how ever sound like your using atom, which is no longer supported by platformio. Use vscode.

wr-rider, 2020-04-22:
Sorry, i thought the error was auto included. i'm using the latest of visual studio code with the latest of python and platformio installed. no matter what version of marlin I've tried it is constantly blue screening my win10. I've uninstalled and reinstalled every thing multiple times including wiping any directory's and such. I've installed a fresh copy of marlin, changed a few lines of code at a time like adding my board, compiled, added drivers, compiled and it will crash at random times compiling. its really fickle on when it crashes. I had been using vs 2.05.2 on a regular bases and even that now blows up. 

Is there any other compiler i can use?
Thanks

ellensp, 2020-04-22:
Still no logs...  Just create a txt log and drag the file onto a reply.  or use pastebin and provide link..  

with a lpc1768/9 you need platformio. so basically no only vscode.

platformio does need to connect to the the internet, in particular http://api.platformio.org/ this is to look up and download the required libraries.

closed 2020-04-27 by thinkyhead