MarlinFirmware/Marlin #17646: BL Touch

Linkeb3, 2020-04-21:
### Bug Description

My BLtouch won't probe more than 9 spots regardless of config, and doesn't create accurate settings as the right side is carving into bed, and left is too high.

### My Configurations

[](https://gist.github.com/Linkeb3/a497ad90f4adecd53c6031fe999223c0)
[](https://gist.github.com/Linkeb3/faee492c414166d28b47cf8470f1386b)

### Steps to Reproduce

I'm using an SKR 1.2, the BLTouch is plugged into z-stop and servos. I was using a BTT35, but disconnected it as a troubleshooting step and plugged in the old LCD.
I've edited the config and advanced config every way i can think of to get the BLtouch to probe more than 9 points - of which it doesn't apparently create a mesh as one side is carving into glass, the other too high. i

**Expected behavior:** create and use a mesh with more than 9 points

**Actual behavior:** poor leveling at 9 points

thierryzoller, 2020-04-21:
Duplicate  of #17399 and #17394 and #17386 and #17349 and #17587 ?

thinkyhead, 2020-04-21:
Closing as a duplicate of these other issues. You must make sure your machine is true and that the probe remains at the same angle to the bed throughout the probing procedure, or the readings will not be useful. One side too high and the other too low is a classic symptom of a mechanical problem.

For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-04-21 by thinkyhead

Linkeb3, 2020-04-21:
this is for marlin 2.0.5.3, the duplicates are not.

Roxy-3D, 2020-04-21:
> this is for marlin 2.0.5.3, the duplicates are not.

More text might help us understand what you are trying to communicate.

thinkyhead, 2020-04-21:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thierryzoller, 2020-04-22:
@thinkyhead Someone did do the testing on what version this was introduced, check this out :  https://github.com/MarlinFirmware/Marlin/issues/17399#issuecomment-616210265

GhostlyCrowd, 2020-04-22:
> @thinkyhead Someone did do the testing on what version this was introduced, check this out : [#17399 (comment)](https://github.com/MarlinFirmware/Marlin/issues/17399#issuecomment-616210265)

I rolled back to that commit and it did not fix the issues for me, i ended up going back to 2.0.3 and when i have time will start walking forward.

 BLtouch going rogue on SKR pro and Mini e3. randomly not deploying probes driving tool head into the earths core, and "black holes" on bed meshes. like the attached image. I don't have anything more to offer with debug probe as were just trying to get things moving we are printing PPE for local medical.  

The image is just an example of the random bad probing when it actually does probe. The black hole is random, appears different spots anywhere on the bed. 

[image]

thinkyhead, 2020-04-23:
> start walking forward.

Walk halfway forward. If it still fails, walk halfway back towards the past, otherwise walk another half way towards now. It will take about 5 tries to find the exact commit.

GhostlyCrowd, 2020-04-23:
> > start walking forward.
> 
> Walk halfway forward. If it still fails, walk halfway back towards the past, otherwise walk another half way towards now. It will take about 5 tries to find the exact commit.

This weekend i hope to try and juggle print jobs and walking back and forth on commits to at least narrow it down. For what its worth its related to https://github.com/MarlinFirmware/Marlin/issues/16169 
https://github.com/MarlinFirmware/Marlin/issues/16795
And
https://github.com/MarlinFirmware/Marlin/issues/17628

All of these behaviours have been experienced personally and it was all solved on my SKR PRO with SD EPEROM emulation. so it appears to be a flash EEPROM emulation related issue. The SKR mini does not have the ability on code to SD EPROM emulate so i cannot speak for that platform.

thinkyhead, 2020-04-23:
A hardware peripheral conflict of some kind is what I'm thinking, and the culprit will be a change somewhere near the HAL. There are continual changes in the peripherals and HALs, so give `bugfix-2.0.x` a check at some point to see if it's better or even just different.

GhostlyCrowd, 2020-04-23:
> A hardware peripheral conflict of some kind is what I'm thinking, and the culprit will be a change somewhere near the HAL. There are continual changes in the peripherals and HALs, so give `bugfix-2.0.x` a check at some point to see if it's better or even just different.

I will keep you posted on this, I had a few friends on SKR Mini platforms were in agreement to this weekend, will do the walk forward and back until we get a narrower field of offending commit.

Linkeb3, 2020-05-02:
Any word on upgrading to the latest?