MarlinFirmware/Marlin #17649: [BUG] .gcode commands executed too fast on BTT SKR Mini 1.4; Ender 3-Pro

LargoW, 2020-04-21:
### Bug Description

.gcode commands are executed way to fast. I did not have this issue with the stock Ender3 PCB (which uses a derived Marlin stock FW). 

Near the end of the print (which was working flawlessly), suddenly the bed gets switched off. After a short while and roughly 2-3 executed gcode (estimation) the hotend switches off, too. The extruder is not switched off, resulting in clogging nozzles. Print must be attended at all times, too, for prevention of damage.

### My Configurations
 - Not required, using stock configuration apart from selecting the SKR Mini IC to build.
 - Can also be replicated by using the released binary file from BigTreeTechs Git repo 
- I use latest Cura for slicing

### Steps to Reproduce

1. Power on printer, goto print from SD Card (it does not matter if using the stock one or a brand one)
2. Select Print (see attached files) and confirm
3. Wait til the last minute

**Expected behavior:** [What you expect to happen]
Temperature should be switched off after print finishes.

**Actual behavior:** [What actually happens]

1. Print runs as expected, heating up bed, then extruder, then printing rim/file as configured
2. Near end of print, roughly during the last minute or so, the bed is switched off
3. Shortly after (less than 30 seconds) the hotend is switched off too
4. Print continues, does not detect cold extruder push neither
4a. Temperature can be set on via LCD back to desired values
4b. Removing the cooldown commands from the gcode fixes this issue

#### Additional Information
Non Working file is PrintTestX.gcode, working file is PrintTestX_NoCool.gcode
Please note that the file was working flawlessly on the ender 3 stock pcb. I have not added any upgrades.
I have a wires attached to the board which may fit a BL Touch at some point in time, but at the moment nothing is connected to these wires. I am using regular end stops for z.

[PrintTestX.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/4512171/PrintTestX.gcode.txt)
[PrintTestX_NoCool.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/4512172/PrintTestX_NoCool.gcode.txt)

ellensp, 2020-04-22:
you need to add a m400 to your end.gcode file before turning off hotends and beds etc. (exact name depends on slicer)

LargoW, 2020-04-22:
Fair enough, is this documented at some point that I have to touch my slicer config? Because I did not need those M400 calls to wait for all gcode to be finished. I knew of the m400 documentation, and as far as I know, Cooldown is part of the regular end procedure not the end-script executed afterwards?
This is especially annoying as all my old pre sliced files, which were working, are now broken.
Is there a workaround for this? Do I need to add the M400 only for temperature or for Z layer as well? Example file only has a single layer, so I haven't checked that and would be pretty annoying to realize this mid print.

randellhodges, 2020-04-22:
What is your `BLOCK_BUFFER_SIZE` from Configuration_adv.h?

Movements are buffered in the planner, temperature changes are processed as soon as they are processed by the serial (not buffered). A really large buffer allows temperature changes to happen much sooner than movements.

So, in a nutshell, you really only need it when you are turning heaters off. So it should go before you do that inside your end gcode. You probably don't need it anywhere else inside the gcode file.

LargoW, 2020-04-22:
Adding M400 to the gcode files works, I can also confirm that there are no issues regarding layer shifts. I was also able to just edit the end script in Cura so I think this can be closed as "not an issue"

closed 2020-04-22 by LargoW