MarlinFirmware/Marlin #17650: [BUG] (short description)

phazei, 2020-04-21:
#SETUP:
SKR v1.4 + TMC2209 w/sensorless homing

### Bug Description

Scenario: Using a z probe on a custom pin for both homing and endstop (not attempting to use sensorless homing for this axis)

The inductive probe is plugged into a custom port (board doesn't support using z-min):
#define Z_MIN_PROBE_PIN P2_00

If only that is defined, then M119 registers z-min registers as always open and z-probe as properly triggered when triggered and open when not triggered.

The issue with that is when homing, it ignores z-probe and continues to move into the head as z-min is open.

So, to fix that the instructions say to enable this:
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

But the probe isn't plugged into the Z_MIN_ENDSTOP_PIN and it now ignores the
#define Z_MIN_PROBE_PIN P2_00

So there needs to be some other option such as
#define Z_MIN_ENDSTOP_USES_Z_MIN_PROBE_PIN

or just a single option such as Z_MIN_PROBE_Z_MIN_ENDSTOP_SHARED which will look at the same pin for them both.

I don't know how to proceed from here.

### My Configurations
config:
https://pastebin.com/Wg9DNLmV
config adv:
https://pastebin.com/uwz5Fk6q

**Expected behavior:** [What you expect to happen]
Homing listens to a probe on a custom z-min port.

**Actual behavior:** [What actually happens]
Probe is ignored and bed crashes

phazei, 2020-04-21:
This is somewhat related to this:
https://github.com/MarlinFirmware/Marlin/issues/15158

thisiskeithb, 2020-04-21:
Have you tried enabling `USE_PROBE_FOR_Z_HOMING`?

phazei, 2020-04-22:
I've been working on it, I haven't used that const specifically, but I did use:
HOMING_Z_WITH_PROBE
Based on a comment from that related issue, and it does seem to have fixed the issue.  I don't know what other things it might affect since there are a lot of conditionals surrounding:
`Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` && `Z_MIN_PROBE_PIN` that are exclusive of each other.  I currently have `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` disabled and am only using:
```
#define HOMING_Z_WITH_PROBE 1
#define Z_MIN_PROBE_PIN P2_00 // Pin 32 is the RAMPS default
```

Would be great if it was confirmed there were no conflicts and added to the docs and to the configuration.h

(I don't see USE_PROBE_FOR_Z_HOMING anywhere in a code search, so I presume you meant the key I used)

thisiskeithb, 2020-04-22:
Update to the latest `bugfix-2.0.x` And see if your issue persists. It has `USE_PROBE_FOR_Z_HOMING`.

phazei, 2020-04-22:
Ah, I'm using the 2.0.5.3.  I'll try that out

phazei, 2020-04-22:
Got it, latest bugfix branch.  The `USE_PROBE_FOR_Z_HOMING` works :)

closed 2020-04-22 by phazei