MarlinFirmware/Marlin #17652: SKR 1.3 will not connect to PC after uploading firmware

messickmatt, 2020-04-22:
# SKR 1.3 will not connect to PC after uploading firmware

### My Configurations
Having the same issue with 2.0.x and bugfix 2.0.x.  Have also tried older releases with the same results.
Board is bigtreetech skr v1.3
Stepper drivers are TMC2208 (UART), although I have this problem regardless of whether they are plugged into the board or not
No LCD screen or other expansions

### Steps to Reproduce
1. Download 2.0.x (or bugfix 2.0.x, same problem)
2. Change platformio.ini environment to LPC1768
3. Configuration.h (only 3 changes)
3.1 change motherboard to BOARD_BTT_SKR_V1_3
3.2 change serial port 1 to -1
3.3 uncomment serial port 2 and set to 0
4. Remove stepper drivers and all jumpers (except usb power) from SKR 1.3
5. Plug board into PC (will recognize and show SD card ONLY if using smoothieware firmware.bin downloaded from bigtreetech github.  Had to copy this to the SD card to unbrick the board after previous attempts to upload Marlin)
6. Run platformio build (runs successfully, one warning about an i2c variable being defined but not used)
7. Platformio upload
8. Push reset button on board

**Expected behavior:** 
Reconnect to PC as COM port and show onboard SD card in file explorer

**Actual behavior:** 
Board connects as COM4.  SD card never appears in file explorer.
This stays the same for 30 seconds, then I get a Windows warning that a USB device has malfunctioned and is not working properly.  COM4 disconnects and the motherboard reconnects as "Unknown USB Device (Device Descriptor Request Failed)"

Opening the SD card, the firmware.bin file has disappeared- only firmware.cur left

#### Additional Information
Have tried several releases of marlin with the same result (2.0.5.3, 2.0.5.2, 2.0.0.0, bugfix 2.0.x)
Have tried with and without stepper drivers attached
Have tried using onboard 5V regulator and USB power only
Have tried another SD card (SanDisk 2GB)

messickmatt, 2020-04-22:
Bed and extruder thermistors are both plugged in

messickmatt, 2020-04-22:
Components plugged in are:
- all motors (except E1)
- Limit switches (XMAX, YMAX, ZMIN inductive probe)
- bed heater (going to SSR)
- Extruder heater
- Hotend fans

messickmatt, 2020-04-22:
just to make sure it wasn't the thermistors, I wired up an extra one I had lying around so all three thermistor ports on the board are now populated (E0, E1, and bed).  The board is still having the same issue.

ellensp, 2020-04-22:
Have you ditched the often provided 128 sd card? as it is crap and corrupts firmware.

ellensp, 2020-04-22:
Also try the known to work firmware.bin here https://github.com/MarlinFirmware/Marlin/issues/17452

swilkens, 2020-04-22:
Could it be that `Platformio upload` (which is essentially a file transfer) isn't yet fully completed before you `Push reset button on board` ?

Try to copy the file to the SD card manually - using a card reader for example - and test again to make sure.

You may also want to format the card properly.

messickmatt, 2020-04-22:
I did try a name brand 2gb as card and it didn't seem to make a difference.
Just confirming- is FAT32 is correct when I reformat it to try again?
I'll try the firmware you sent when I get off work

messickmatt, 2020-04-23:
I tried your compiled firmware with a reformatted 2GB SanDisk card, and I am still having the same issue.  What hardware configuration does this firmware correspond to, and is there anything that needs to be plugged into the board for the firmware to work?  Do I need the stepper drivers/UART jumpers connected?
I'm still confused because the board appears to connect fine when using the preconfigured smoothieware from BigTreeTech:  https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/tree/master/BTT%20SKR%20V1.3/firmware/smoothieware%2BTFT35

messickmatt, 2020-04-23:
Issue is resolved.
Problem was a faulty extruder thermistor at 0 ohms
thanks @thisiskeithb

closed 2020-04-23 by messickmatt