MarlinFirmware/Marlin #17654: [FR] Reactivate tmcstepper libraries for stm32 boards in platformio.ini again

bergyla, 2020-04-22:
Incompatibility issues between stm32 and tmcstepper libraries were fixed several months ago.
However tmcstepper is disabled in most of stm32 board entries inside the platformio.ini.

As a newbie I was wondering while still calling the correct (tmcstepper version) in the common section of the platformio.ini the compile still failed with blaming tmcstepper.
I had to learn that there is a libignore entry in specific board section and tmcstepper is added here by default...

As I could see there are many others like me...

rjacobs1969, 2020-04-23:
I second this...It took me hours to figure that out...

PD  the adafruit neopixel library as well, it also has been fixed for the stm32 months ago

sjasonsmith, 2020-05-02:
@bergyla and @rjacobs1969, can you be more specific about which boards you are having issues with?

TMCStepper is already enabled for most STM32 boards. Sometimes a particular board may not be updated because nobody using that board is contributing pull requests. STM32 boards often require some work to properly select which timer to use (at least for UART-based drivers), so I usually only update boards I can actually verify changes on.

If you have it working on a board for which it is currently ignored, please create a pull request to update the platformio.ini file!

bergyla, 2020-05-04:
As far I know, the root cause for disabling TMCStepper from STM32 platform was based on some incompatibility to the STM core library and should be fixed for all stm32 derivatives.
Currently the following boards still disable tmcstepper:
STM32F4, STM32F7, mks_robin_pro, STM32F103CB_malyan, STM32F401VE_STEVAL, FLYF407ZG, STM32F407VE_black

For  creating a Pull request, I must carefully learn how to do that the right way...

Regards

sjasonsmith, 2020-05-07:
I've just spent some time increasing the consistency of some STM32  environments, so thought through all of these.

> STM32F4, STM32F7
These aren't actually usable as far as I know. I believe they are just dummy environments that build against that HAL.

> STM32F103CB_malyan
I be4lieve this has built-in stepper drivers, so there would be no reason to support TMCStepper.

> STM32F401VE_STEVAL, STM32F407VE_black
These are development boards with no printer-specific functionality. If someone wants to use a TMC stepper with them, it seems reasonable to expect them to add it themselves.

> FLYF407ZG
I added this in my PR #17904 if it gets it in. There is some chance this introduces a timer conflict that could make these boards unusable, but if anyone is using these boards they could let us know then.

> mks_robin_pro
This one's reasonable. My current PR isn't dealing with boards using the STM32F1 HAL. It looks like MKS actually enables it in their repo with software serial on timer number 4, so it should probably be safe to add it here.

sjasonsmith, 2020-06-25:
I don't think there is any more to be done for this. If there are remaining gaps _that actually impact users_ they should be reported as separate issues.

closed 2020-06-25 by sjasonsmith