MarlinFirmware/Marlin #17655: [BUG] power loss recovery not working with latest bugfix

drfil120, 2020-04-22:
### Bug Description
Power loss recovery not working with latest Marlin bugfix (downloaded this morning)
Not working anymore since renaming "power_loss_recover.cpp" to "powerloss.cpp
Board : BTT skr mini E3 v1.2 + mini UPS 24V
With M43 pins debug the PIN PC12(44) indicates "output".
If i set the pin with M42 to input powerloss works and my heatbed raises.

### My Configurations
[configurations.zip](https://github.com/MarlinFirmware/Marlin/files/4513956/configurations.zip)

### Steps to Reproduce
1. [First Step]
Compile latest Marlin bugfix
2. [Second Step]
Cut power to printer and bed must raise

**Expected behavior:** [What you expect to happen]
Bed raises after power cut, screen indicates power cut

**Actual behavior:** [What actually happens]
no reaction from printer, no indication on screen

#### Additional Information

[image]
[image]

boelle, 2020-06-20:
@drfil120 is the issue still there?

drfil120, 2020-06-21:
The issue is solved.
Thanks

closed 2020-06-21 by drfil120