MarlinFirmware/Marlin #17656: [BUG] AUTO_BED_LEVELING_UBL doesn't use MIN_PROBE_EDGE

compunited, 2020-04-22:
AUTO_BED_LEVELING_UBL doesn't use MIN_PROBE_EDGE. It goes outside the allowed radius which leads to serious problems on delta. (touching clamps, hitting mechanical end stops,...)

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

compunited, 2020-06-24:
You're not serious? After 60 days you had a look at it and close it after 3 days?

boelle, 2020-06-24:
yes serious, 3 days should be enough to report back

sjasonsmith, 2020-06-24:
I have a delta I could try to reproduce it on, but I’ll need your configs as a starting point.

closed 2020-08-20 by github-actions[bot]