MarlinFirmware/Marlin #17658: [FR] (Conditional Autostart)

TobiasHullette, 2020-04-22:
Hello,

I am working on an odd project. I am making an Automatic Chicken Door using an old 3D printer motherboard, LCD, and stepper motor. I know that I'm taking a long way around using Marlin, but I have learned a lot and am now almost done.

My plan is to connect the motherboard to a programmable timer, so that at 6am, the system boots, and automatically runs the GCODE to turn the motor and open the door. I have that working by using Marlin's AutoStart feature. However, I don't want to run the same GCODE at 9pm when I want to close the door.

Is there anyway to code something that would do something like:

IF ZMAX triggered, run gcode #10074 
Else run gcode #2

?

Otherwise im going to have to run a gcode that opens the door, waits all day and then closes. But I don't want everything to be using power uselessly.

tomarigr, 2020-05-02:
I originally opened this issue because I mistakenly read the title "conditional astronaut" and I said wtf! 
But this is even more weird! Man, you are using a nuclear bomb to kill just one little spider! I suggest you use the Arduino environment and code just a few lines of code and do it nice and clean. Nothing to do with Marlin!

TobiasHullette, 2020-05-04:
> I originally opened this issue because I mistakenly read the title "conditional astronaut" and I said wtf!
> But this is even more weird! Man, you are using a nuclear bomb to kill just one little [...]

I know it is stupid. But I would really like to make this work. Do you know how to?

tomarigr, 2020-05-04:
Well, I would set up an Arduino connected to the serial port of marlin board and send gcode accordingly instead of messing with marlin code. God(devs) knows where to start in like a gazillion lines of code and what else would break if you tamper with it! Even better you can use an esp32 and command it(the marlin board) over the internet from the other side of the world. Power it on and off whenever you want etc. 

But this GitHub rep is not for this kind of help. You are better off to the Arduino/esp33 forums

charlespick, 2021-05-19:
I agree, I think this kind of functionality is much better for a host software instead.

thisiskeithb, 2021-09-24:
You can now assign G-code to buttons/pins which should cover this FR:

https://github.com/MarlinFirmware/Marlin/blob/e85f6225053de257ee6f2652f097d0939d0dbde7/Marlin/Configuration_adv.h#L3791-L3820

closed 2021-09-24 by thisiskeithb