MarlinFirmware/Marlin #17660: Incorrect output on M503

token47, 2020-04-22:
Using Marlin 2.0.5.3 on atmega2560.

When connecting to the printer, a list of values on the EEPROM is listed (the same list you get with issuing M503 at any time).

Problem: some lines are not preceeded with "echo:" (specially the ones for second extruder in some cases -- but not all of them). This is bad because if you don't have "verbose" mode on (in Simplify3D for example) you only see what the echo commands shows and you miss the others. If you back up your configuration copying and pasting (before an upgrade, for example) you will miss some entries.

Example:

```
Connecting...
start
Printer is now online.
echo:Marlin 2.0.5.3
echo: Last Updated: 2020-03-31 | Author: (token47, Tevo Tarantula Pro)
echo:Compiled: Apr 22 2020
echo: Free Memory: 1990  PlannerBufferBytes: 1648
echo:V76 stored settings retrieved (660 bytes; crc 11821)
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:; Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 T1 D1.75
echo:  M200 D0
echo:; Steps per unit:
echo: M92 X80.06 Y80.06 Z399.29
echo: M92 T0 E408.00                             <------------- these are OK
echo: M92 T1 E408.00                             <------------- 
echo:; Maximum feedrates (units/s):
echo:  M203 X500.00 Y500.00 Z12.00
echo:  M203 T0 E70.00                            <------------- OK for T0
  M203 T1 E45.00                                 <------------- missing 'echo' for T1
echo:; Maximum Acceleration (units/s2):
echo:  M201 X6000.00 Y6000.00 Z200.00
echo:  M201 T0 E10000.00                         <------------- OK for T0
  M201 T1 E10000.00                              <------------- missing 'echo' for T1
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1500.00 R1500.00 T1500.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Auto Bed Leveling:
echo:  M420 S0 Z2.00
echo:  G29 W I0 J0 Z-0.09286
echo:  G29 W I1 J0 Z0.04990
echo:  G29 W I2 J0 Z0.13755
echo:  G29 W I0 J1 Z-0.07533
echo:  G29 W I1 J1 Z0.00482
echo:  G29 W I2 J1 Z-0.05780
echo:  G29 W I0 J2 Z0.25025
echo:  G29 W I1 J2 Z0.17512
echo:  G29 W I2 J2 Z0.05741
echo:; Material heatup parameters:
echo:  M145 S0 H200 B60 F0
echo:  M145 S1 H250 B75 F0
echo:; PID settings:
echo:  M301 P20.41 I1.86 D56.00
echo:; LCD Contrast:
echo:  M250 C195
echo:; Z-Probe Offset (mm):
echo:  M851 X0.00 Y15.00 Z-1.40
echo:; Stepper driver current:
echo:  M906 X600 Y600 Z600
echo:  M906 T0 E600                              <------------- these are OK
echo:  M906 T1 E600                              <-------------
echo:; Driver stepping mode:
echo:; Linear Advance:
echo:  M900 T0 K0.00                             <------------- OK for T0
  M900 T1 K0.00                                  <------------- missing 'echo' for T1
echo:; Filament load/unload lengths:
echo:  M603 T0 L0.00 U0.00                       <------------- these are OK 
echo:  M603 T1 L0.00 U0.00                       <-------------
echo:; Tool-changing:
echo: Z0.00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
Testing E1 connection... OK
echo:SD card ok
echo:Settings Stored (660 bytes; crc 8156)
```

seitmand, 2020-04-24:
I have noticed something similar.  Using Arduino 1.8.12 and Marlin 2.0.4.4, I do not see any missing "echo" commands.
 
On start-up or on **M503 S1** :
```
echo:Marlin 2.0.4.4

echo: Last Updated: 2020-02-27 | Author: 3DMakerWorld,Seitman
echo:Compiled: Apr 23 2020
echo: Free Memory: 2997  PlannerBufferBytes: 1504
echo:SD card ok
echo:V76 stored settings retrieved (687 bytes; crc 4965)
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius

echo:; Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 T1 D1.75
echo:  M200 D0
echo:; Steps per unit:
echo: M92 X80.21 Y80.21 Z1511.90 E473.20
echo:; Maximum feedrates (units/s):
echo:  M203 X500.00 Y500.00 Z4.00 E150.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X6000.00 Y6000.00 Z100.00 E10000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 B20000.00 S0.00 T0.00 X20.00 Y20.00 Z0.40 E5.00
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Hotend offsets:
echo:  M218 T1 X0.00 Y0.00 Z0.000
echo:; Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:; PID settings:
echo:  M301 E0 P22.61 I1.69 D75.95
echo:  M301 E1 P22.61 I1.69 D75.95
echo:; Tool-changing:
echo: Z2.00
echo:SD card ok

```
And on using **M503 S0** :

```
  M200 D1.75
  M200 T1 D1.75
  M200 D0
  M92 X80.21 Y80.21 Z1511.90 E473.20
  M203 X500.00 Y500.00 Z4.00 E150.00
  M201 X6000.00 Y6000.00 Z100.00 E10000.00
  M204 P3000.00 R3000.00 T3000.00
  M205 B20000.00 S0.00 T0.00 X20.00 Y20.00 Z0.40 E5.00
  M206 X0.00 Y0.00 Z0.00
  M218 T1 X0.00 Y0.00 Z0.000
  M145 S0 H180 B70 F0
  M145 S1 H240 B110 F0
  M301 E0 P22.61 I1.69 D75.95
  M301 E1 P22.61 I1.69 D75.95
 Z2.00

```
However, the error existed in Marlin 1.1.3, albeit it was a little different than the error presented by 

> token47

Using **M503 S1** :
```
start
echo:Marlin 1.1.3

echo: Last Updated: 2017-06-04 12:00 | Author: 3DMakerWorld,Seitman
Compiled: Apr 23 2020
echo: Free Memory: 3086  PlannerBufferBytes: 1296
echo:EEPROM version mismatch (EEPROM=V76 Marlin=V38)
echo:Hardcoded Default Settings Loaded
echo:  G21    ; Units in mm
echo:  M149 C ; Units in Celsius

echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 T1 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X80.21 Y80.21 Z1511.90 E473.20
echo:Maximum feedrates (units/s):
echo:  M203 X500.00 Y500.00 Z10.00 E150.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X9000 Y9000 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Hotend offsets:
echo:  M218 T1 X45.72 Y1.68
echo:Material heatup parameters:
echo:  M145 S0 H185 B70 F255
  M145 S1 H230 B100 F255       <-----  missing "echo" but output is correct below
echo:PID settings:
echo:  M301 P8.61 I2.04 D79.95   <-----missing E0 and E1 values even when these values differ
echo:  M304 P372.36 I30.10 D1151.54
echo:SD card ok
echo:SD card ok

```  
But using **M503 S0** the missing values for M304 E1 are present
```
 G21    ; Units in mm
  M149 C ; Units in Celsius

  M200 D1.75
  M200 T1 D1.75
  M200 D0
  M92 X80.21 Y80.21 Z1511.90 E473.20
  M203 X500.00 Y500.00 Z10.00 E150.00
  M201 X9000 Y9000 Z100 E10000
  M204 P3000.00 R3000.00 T3000.00
  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
  M206 X0.00 Y0.00 Z0.00
  M218 T1 X45.72 Y1.68
  M145 S0 H185 B70 F255
  M145 S1 H230 B100 F255
  M301 E0 P8.61 I2.04 D79.95
  M301 E1 P8.61 I2.04 D79.95     <-----  but the values are really present
  M304 P372.36 I30.10 D1151.54

```

ellensp, 2020-04-26:
This has been fixed in bugfix, please close issue.

seitmand, 2020-04-26:
@ellensp I would close this issue but I do not have the "close issue" button.

PR #17665 (2020-04-25): Fix 17660 Inconsistent M503 output with DISTINCT_E_FACTORS
### Description

See issue #17660 
Inconsistent M503 output with DISTINCT_E_FACTORS

### Benefits

M503 output is consistent with DISTINCT_E_FACTORS

### Related Issues

Fix #17660
closed 2020-04-26 by token47