MarlinFirmware/Marlin #17666: [BUG] Raising flowrate with TMC2208 drivers results in unexpected behaviour

Ohmarinus, 2020-04-23:
### Bug Description

Raising the flowrate to 200% during a print causes the machine to vibrate heavily and make really loud noises, see the video here:
https://www.youtube.com/watch?v=ZZZDV9c6sxI

### My Configurations

Using Marlin 2.0.5.2
[Marlin 2.0.5.2.zip](https://github.com/MarlinFirmware/Marlin/files/4521101/Marlin.2.0.5.2.zip)

### Steps to Reproduce

1. I'm building a pen plotter, DIY design, it was using strings and is now converted to belts so the pulley size increased and I increased the steps/mm from 36.57 to 100 which is the correct setting.

2. During the first test print I noticed the machine suddenly printed much slower than before (I thought this was weird, changing the pulley size and raising the steps/mm accordingly shouldn't slow down a print)

3. I raised the flow rate to test out how fast the machine could draw while maintaining dimensional accuracy and now have any ringing.

**Expected behavior:** 
I expected the printer to speed up linearly with the speed settings while maintaining silent operation like I am used to from the TMC2208 drivers.

**Actual behavior:**
The machine did speed up, but not by 200% and it started making really bad noises (see video).

#### Additional Information

Here again is the video:
https://www.youtube.com/watch?v=ZZZDV9c6sxI

What I did:
- Disable HYBRID_THRESHOLD because the machine runs above 100mm/s almost all of the time and stealthchop doesn't lose steps anyway because the x-carriage is superlight.
- Check the TMC connection, all motors are reporting back with a good connection
- Check if the EEPROM was correctly updated after uploading new firmware for the new steps/mm
- Check the GCODE file to see if the F-speeds were still set correctly, which they were

I hope anyone recognizes what's going on. This pen plotter was built to be fast so I am really curious to find out what could cause the problem. Is it raising flow rate that causes an issue? Or is there something else I'm overlooking?

In the end, despite the noises, the images come out perfectly so there are no artifacts visible of the noise that is generated in the line-art. However, I don't believe this can be good for the machine. It sounds as if it is destroying itself slowly.

boelle, 2020-06-20:
@Ohmarinus is the issue still there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle