MarlinFirmware/Marlin #17668: [BUG] UBL edit saves value to wrong location

mobri2a, 2020-04-23:
### Bug Description

After running G29 P1 on a clean EEPROM I used G29 P4 to edit one of the leveling points. When saved the value appears in a completely different location. Ex. in a 5x5 grid with close/left as 0,0, I adjusted row 2, column 3. It showed current value as -0.067, which was odd because it should have been +0.142. I adjusted it, but had to go all the way up to +0.745. When I saved the adjustment the new value appears in row 2 column *1*, not column *3*. 

Also when printing the far right column the printer slams the hot end into the bed when at the far edge. Strange because this is the high edge of the bed and the printer is lowering the head significantly, as if the left side values are being used instead of right. This looks like an indexing problem on the x and y arrays used to store/retrieve the mesh. One of them is flipped/mirrored on the y axis, perhaps.

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/4522276/Configurations.zip)

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Perform steps to create the UBL mesh automatically
2. Output the mesh values with G29 T
3. Pick a point on the right side of the bed and use G29 P4 to edit
4. Note the value returned for the closest point does not match what is supposed to be there. Find the value in the mesh output from step 2 - it's coming from another location (row 2, column 1 in my case).
5. Adjust the current point and save the setting
6. G29 T - note that the new value is not stored in the correct location and is quite higher than the probed value.

**Expected behavior:** When manually editing a point in the UBL mesh it should return the value for the point being edited and save the adjustment to the correct point. The adjusted value should be reasonably close to the probed value, not 0.5mm different.

**Actual behavior:** Mesh edit is reading current value from the wrong point and saving new value to the same wrong point. Adjusted value is unreasonably different than expected, in addition to being stored in the wrong location.

#### Additional Information

After G29 P1 T:
```
Recv: Bed Topography Report:
Recv: 
Recv:     ( 25,195)                      (195,195)
Recv:         0       1       2       3       4
Recv:  4 | +0.047  +0.047  +0.080  +0.162  +0.185
Recv:    |
Recv:  3 | -0.015  -0.015  +0.040  +0.153  +0.207
Recv:    |
Recv:  2 | -0.067  -0.067 [+0.005] +0.142  +0.252
Recv:    |
Recv:  1 | -0.090  -0.090  -0.002  +0.167  +0.362
Recv:    |
Recv:  0 | -0.085  -0.085  +0.015  +0.220  +0.425
Recv:         0       1       2       3       4
Recv:     ( 25, 25)                      (195, 25)
```

G26 test shows these values reasonably close. I edited point in row 2, column 3 (nozzle is over this point). LCD displays current value -0.067. Nozzle is buried in the bed surface. Adjusted to +0.745. Now G29 T shows:

```
Recv: Bed Topography Report:
Recv: 
Recv:     ( 25,195)                      (195,195)
Recv:         0       1       2       3       4
Recv:  4 | +0.047  +0.047  +0.080  +0.162  +0.185
Recv:    |
Recv:  3 | -0.015  -0.015  +0.040  +0.153  +0.207
Recv:    |
Recv:  2 | -0.067  +0.745 [+0.005] +0.142  +0.252
Recv:    |
Recv:  1 | -0.090  -0.090  -0.002  +0.167  +0.362
Recv:    |
Recv:  0 | -0.085  -0.085  +0.015  +0.220  +0.425
Recv:         0       1       2       3       4
Recv:     ( 25, 25)                      (195, 25)
```

After resetting the printer I did another test, loading mesh from storage back to first display above. I positioned the nozzle over row 2, col 4. Value displayed is -0.068. Had to adjust to 0.925 to get the nozzle to proper height (note: this is insanely high - 0.5mm higher than probed). Value is saved to row 2 col 1 again.

```
Recv: Bed Topography Report:
Recv: 
Recv:     ( 25,195)                      (195,195)
Recv:         0       1       2       3       4
Recv:  4 | +0.047  +0.047  +0.080  +0.162  +0.185
Recv:    |
Recv:  3 | -0.015  -0.015  +0.040  +0.153  +0.207
Recv:    |
Recv:  2 | -0.067  +0.925 [+0.005] +0.142  +0.252
Recv:    |
Recv:  1 | -0.090  -0.090  -0.002  +0.167  +0.362
Recv:    |
Recv:  0 | -0.085  -0.085  +0.015  +0.220  +0.425
Recv:         0       1       2       3       4
Recv:     ( 25, 25)                      (195, 25)
```

mobri2a, 2020-04-23:
I tried another test on row 4 col 0. Value returned on edit closest: -0.068! Here's the result after adjusting it. Again stored in row 2 col 1, and again way higher than +0.047, which was a reasonably accurate measurement. 
```
Recv: Bed Topography Report:
Recv: 
Recv:     ( 25,195)                      (195,195)
Recv:         0       1       2       3       4
Recv:  4 | +0.047  +0.047  +0.080  +0.162  +0.185
Recv:    |
Recv:  3 | -0.015  -0.015  +0.040  +0.153  +0.207
Recv:    |
Recv:  2 | -0.067  +0.505 [+0.005] +0.142  +0.252
Recv:    |
Recv:  1 | -0.090  -0.090  -0.002  +0.167  +0.362
Recv:    |
Recv:  0 | -0.085  -0.085  +0.015  +0.220  +0.425
Recv:         0       1       2       3       4
Recv:     ( 25, 25)                      (195, 25)
```

mobri2a, 2020-04-23:
Also tried this on the bugfix release with the same results.

boelle, 2020-06-20:
@mobri2a is the issue still there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle