MarlinFirmware/Marlin #17671: [BUG] Units halved when powering drivers after boot

MasterPuffin, 2020-04-23:
### Bug Description
I have set up my Octoprint on an Raspberry Pi 3B. I use a relay and the PSU [control plugin](https://github.com/kantlivelong/OctoPrint-PSUControl) to control the power to my printers power supply. The RPi is delivering power to the mainboard of the printer via USB. My printer is an Creality Ender 3 with a MKS Gen L V2.0 Board and TMC2209 stepper drivers running in UART mode.

Now I ran into the following bug: Octoprint is connected to the printer, as is connects on startup. When I now power on the printer using the PSU control plugin all measurement units are halved. This means when I / gcode tells the printer to move 10 units, it actually moves 20 but reports at being in position it would have been if it had been moving 10 units.

I found a workaround to this problem using the following:
Octoprint is connected to the printer as above. Now turn on the PSU via the plugin, disconnect Octoprint from the printer then reconnect it wich in turn reboots Marlin. Using this the measurements are correct.

So
`Connected -> PWR on`
results in halved measurement units
but
`Connected ->PWR on -> Disconnect -> Connect`
works

I don't know if this is actually an issue with Octoprint or Marlin, so I posted this issue to both, see the corresponding ticket _will update once online_.

### My Configurations

Available at https://github.com/MasterPuffin/MKS-GEN-L-V2-Marlin-Firmware-for-Ender-3

### Steps to Reproduce

**Expected behavior:**
When I move the printer head by 10 units (for example on the X axis) starting at (0|0|0) it will be at (10|0|0) and report being at (10|0|0).

**Actual behavior:**
The printer moved to (20|0|0) but reported being at (10|0|0).

sjasonsmith, 2020-05-29:
This is a common issue when boards are powered over USB. The drivers aren't powered up, so they can't receive the configuration that is sent to them over serial. The most obvious solution is to prevent your board from being powered over USB! There are a variety of methods, from modifying cables to re-configuring the hub on your raspberry pi. If you Google around you should find some options.

Your disconnect/reconnect workaround is a good one. It works for AVR boards because connecting over serial triggers a reboot. It won't work for most or all 32-bit boards.

Another alternative is to enable the `PSU_CONTROL` feature. This will reconfigure your drivers when you turn on your power supply with the `M80` command. It doesn't really matter if you have a controllable power supply, just assign it an unused pin and let it pretend it has a power supply to control! The downside of this is that you will have to send that M80 command every time you turn on your printer.

Since this isn't really a bug in Marlin, I'd suggest you close this issue, and head over to Discord or one of the other user forums to help you figure out a good way to proceed with this.

boelle, 2020-06-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-21 by boelle