MarlinFirmware/Marlin #17672: lcd_pause_show_message causing compilation errors

mrseeker, 2020-04-23:
https://github.com/MarlinFirmware/Marlin/blob/0518dec60d0931745efa2812fa388f33d68cfa29/Marlin/src/lcd/ultralcd.cpp#L1526

I am currently running Marlin 2.0.5.3, but when I turn the adv_config ADVANCED_PARK_FEATURE on, I get an error from this line saying that lcd_pause_show_message is undefined. Commenting this out does seem to work and I can compile everything properly.

sjasonsmith, 2020-05-02:
Please attach your configuration files as requested by the issue template. This makes it much easier for anyone trying to help to reproduce the issue you are reporting.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle