MarlinFirmware/Marlin #17675: [BUG] Filament runout detection

Exalik, 2020-04-23:
### Bug Description

The filament runout detection does not make a break of/stop the printer.

### My Configurations
[Marlin-Bugfix-2.0.x Config.zip](https://github.com/MarlinFirmware/Marlin/files/4523996/Marlin-Bugfix-2.0.x.Config.zip)
Bugfix-2.0.x downloaded on the 04/22/2020 at 04∶38∶21 CEST

### Steps to Reproduce

Launch a print and wait of end of filament or simulate a runout of filament.

**Expected behavior:** The printer should make a break to change the filament.

**Actual behavior:** The printer continue to print.

#### Additional Information

Geeetech A10M with TMC2208

boelle, 2020-06-20:
@Exalik is the issue still there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle

shitcreek, 2020-07-03:
This is probably the same issue as this: https://github.com/MarlinFirmware/Marlin/issues/16862