MarlinFirmware/Marlin #17677: [BUG] M261 not Working on BTT SKR 1.3

robin-gdwl, 2020-04-23:
Similar issue as [#15754], thought it was an issue on my end but since others have this problem it might be more than that. 

I have an Arduino Nano (Mega328P) that I want to use as an i2c Slave to receive and send(on request) data to a Bigtreetech SKR 1.3 (LPC1768).
My I2C pins are P0.0 and P0.1 usually used for E1STP and E1DIR. 
If I send a command to the slave with M260 The data is received by the slave. 
**If I try to send a request with `M261 A9 B3` nothing is acknowledged by the slave.**
This is the serial response from the SKR Board (Debug with `M111 S2`):
```
feeder> M261 A9 B3
TWIBus::address: 9
TWIBus::relay: 3
TWIBus::request: 3
TWIBus::request fail: 9
ok
```
This is (a simplified version of) the request handler running on the Arduino:  
```
void requestEvent() { 
  byte response[3]; 
  for (byte i=0;i<3;i++) { 
    response[i] = (byte)answer.charAt(i); 
  }
  Wire.write(response,sizeof(response));
  Serial.println("Request event");
  delay(1000);                       // wait for a second
}
```
What could be the problem here? requesting Data from the arduino slave with another arduino nano as master works fine ([code here](https://piandmore.wordpress.com/2016/09/13/arduino-nano-i2c-master/)). Is there something I am missing or is this a bug?

ellensp, 2020-04-23:
Did you add the i2c pull up resistors anywhere? 
https://learn.sparkfun.com/tutorials/i2c/i2c-at-the-hardware-level

robin-gdwl, 2020-04-23:
> Did you add the i2c pull up resistors anywhere?
> https://learn.sparkfun.com/tutorials/i2c/i2c-at-the-hardware-level

No I did not. is a pullup resistor necessary for requesting but not for sending?

ellensp, 2020-04-23:
As I understand it the 328p i2c lines are not open drain but standard io, so it will receive data.
But the lpc1768 is real open drain i2c so need the pullups to receive a high bits

robin-gdwl, 2020-04-23:
> As I understand it the 328p i2c lines are not open drain but standard io, so it will receive data.
> But the lpc1768 is real open drain i2c so need the pullups to receive a high bits

Sounds plausible. I will implement a pullup resistor and report if it works. 
Thank youfor the quick help.

robin-gdwl, 2020-04-23:
I added a 4.7K Pullup Resistor, however there is no change in behaviour. 
The requestEvent on the arduino is never triggered so it is not only, that the LPC1768 has problems receiving it does not seem to ever send the request properly (from my limited understanding).

randellhodges, 2020-04-23:
See
https://github.com/MarlinFirmware/Marlin/issues/15754#issuecomment-618558127

robin-gdwl, 2020-04-23:
> See
> [#15754 (comment)](https://github.com/MarlinFirmware/Marlin/issues/15754#issuecomment-618558127)

This Works in combination with the pullup-resistors !! 
Thank you very much.

robin-gdwl, 2020-04-23:
> For LPC see:
> [#15754 (comment)](https://github.com/MarlinFirmware/Marlin/issues/15754#issuecomment-618558127)

Is this solution already documented somewhere? I  just saw that this is implemented in [macros.h](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/core/macros.h#L295), is could this be implemented to automatically activate it ?

closed 2020-04-23 by robin-gdwl

randellhodges, 2020-04-23:
I happened to be doing other I2C work on LPC and happened across this exact issue (I2C for EEPROM).

I just finished submitting a PR to fix it for M261.