MarlinFirmware/Marlin #17685: [BUG] z-probe-offset doesn't always move print head

phazei, 2020-04-24:
About half the time when changing the z-probe-offset, it doesn't actually move.  Or it might start moving in one direction, then doesn't go back.  Need to reset and home again to get it to start working.  Bugfix 2.0 branch

configs:
https://pastebin.com/dnbbkuXa
https://pastebin.com/GL9Gr6fA

thisiskeithb, 2020-04-24:
Have you tried enabling `INTEGRATED_BABYSTEPPING`?

phazei, 2020-04-24:
I'll try that out.  I merged in today's copy of bugfix, but it didn't build so I reverted back a few commits.
I did notice when I turn the nob to adjust the z-probe-offset, I hear a very quiet frequency go for a few moments, but nothing moves at all.

phazei, 2020-04-24:
Haven't done it yet, but I found this and believe it to be related:
https://github.com/MarlinFirmware/Marlin/issues/16828

I tried the integrated baby steps, I read more about the baby stepping too.  Disabled it on xy, and enabled the graphic overlay, which changed the z-probe-offset screen, I like the graphic, makes it more clear.

```
#if ENABLED(BABYSTEPPING)
  #define INTEGRATED_BABYSTEPPING
  //#define BABYSTEP_WITHOUT_HOMING
  //#define BABYSTEP_XY
  #define BABYSTEP_INVERT_Z false
  #define BABYSTEP_MULTIPLICATOR_Z  64
  //#define BABYSTEP_MULTIPLICATOR_XY 1

  #define DOUBLECLICK_FOR_Z_BABYSTEPPING
  #if ENABLED(DOUBLECLICK_FOR_Z_BABYSTEPPING)
    #define DOUBLECLICK_MAX_INTERVAL 1250
    //#define BABYSTEP_ALWAYS_AVAILABLE
    //#define MOVE_Z_WHEN_IDLE
  #endif

  //#define BABYSTEP_DISPLAY_TOTAL
  #define BABYSTEP_ZPROBE_OFFSET
  #if ENABLED(BABYSTEP_ZPROBE_OFFSET)
    //#define BABYSTEP_HOTEND_Z_OFFSET
    #define BABYSTEP_ZPROBE_GFX_OVERLAY
  #endif
#endif

```
I disabled BS xy, changed multiplier to 64 since z steps/mm=6400.  Also disabled software z-min endstop.

Reset board, auto-homed.  z gets set to 3.350.  Set z=0, paper test loose.  I'm able to go to Move Z - 0.025mm and roll the nob up and down around 0mm as much as I'd like and it remains consistent, good paper test at -0.050mm.  Set back to z=0.  Go to adjust z-probe-offset, it starts at -3.350, no movement at all.

Reset, home, set z=0, don't roll the nob a lot.  go to z-probe-offset, at -3.350. I move it to -3.400, just right, move back to -3.350, doesn't change, back, move to -3.400, just gets tighter.  Only goes down, won't go up

phazei, 2020-04-24:
So, more insight, if I roll z-probe-offset around enough, it stops working entirely, in either direction.

But, I noticed z-probe-offset actually does go up, but significantly less than when it goes down.
I moved it from -3.350 to -3.400, paper went from loose to just right.  Then I had to move it from -3.400 all the way to -3.060 for it to be loose again.

phazei, 2020-04-24:
I read if the babystep multiplier was too high it could make a difference.  Tried it at 1, omg, 6 or 7 clicks of the dial to go from 3.350 to 3.351.  z-probe-offset did move a bit, but then stopped entirely.  I had to really crank that nob to get anywhere noticeable at that small a step, going to move it back to 64, at least that only was 1 click per 1/1000

phazei, 2020-04-24:
So, BS Z multiplier back to 64.
I tried turning off stealthchop for the z-axis, wow, didn't realize how much louder that is.

So, z-probe-offset is still going down at what seems to match the nobs amount, but up incredibly slowly.  When it stops working, I usually reset.  I believe homing makes it work again as well.
One thing really odd, was after I hadn't reset it for a bit, and just played with moving the z-axis in the Move Z menu, I went back to z-probe-offset, and then, it would only move in the up direction now, and not at all in the down :O  so it switched which was was working, which is really odd....

phazei, 2020-04-24:
videos! see, I'm not seeing things
You can see the tiny black screw on the connector drifting to the left but barely any to the right:
https://photos.app.goo.gl/aD7sdphwsqnSK7kN6

Here you can see it drifting slowly right, but none to the left:
https://photos.app.goo.gl/6TcitsRAapzirKNd8

Same build of fw, no settings changed, stealthchop off.

I'm not sure if this is all a single issue, or multiple.

1) It only goes in one direction
2) It stops working entirely after a bit.

phazei, 2020-04-24:
I also tried adjusting it during a print, it only goes one way there too, so can't really be used to tune anything.  Also double pressing the spin wheel on the info screen does not take me to the z offset page.

hamster65, 2020-04-25:
I am using the current bugfix and was adjusting the z offset with babystepping. My impression was that the z stepper sometimes did not react when I turned the encoder.
In addition, it was nearly impossible to set the correct z offset. Sometimes, I endet up with +1.2, on the next print, the nozzle was way to high and I had to adjust it to -0.6.

phazei, 2020-04-26:
@hamster65

I just found a ticket with a solution that seems to work for me:
https://github.com/MarlinFirmware/Marlin/issues/17323

hamster65, 2020-05-01:
With setting both pre and post MINIMUM_STEPPER_DIR_DELAY to 100, baby stepping seams to work fine.

pocharlies, 2020-05-23:
i had the same problem, SKR 1.4 turbo with TMC 2209 and my babysteeeps are not working..

I set it to MINIMUM_STEPPER_DIR_DELAY to 60 and it works perfectly.

On the other hand, this couldn't affect to the prints precision if it's set to 20?

chandigital, 2020-06-18:
My SKR 1.3 with TMC 2208 (UART) was consistently losing ability to do Z offset adjustments from the knob (BTT TFT 3.5 V3). With no MINIMUM_STEPPER_{PRE,POST}_DIR_DELAY set, the stepper would consistently stop working between -3.3 to -3.4. Defining MINIMUM_STEPPER_{PRE,POST}_DIR_DELAY to 100 also *failed* (although I was able to go a bit further, I think I got to about -5.8 before heading in the opposite direction and shortly afterwards having the stepper freeze up). However, a delay value of 200 seems to work for me.

One more thing that I noticed is that if I left the menu on the Adjust Z Offset screen (showing current adjustment value) for more than a minute or so that any additional adjustments would stop working.  Not sure if this tidbit adds any value.

boelle, 2020-06-21:
@phazei still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle