MarlinFirmware/Marlin #17687: [BUG] Sensorless homing sensitivity issues with Quick_Home enabled.

jreis1, 2020-04-24:
### Bug Description
Sensorless homing sensitivity issues with Quick_Home enabled. 

With sensorless homing active, when Quick_Home **is not enabled**, both X and Y axis home smoothly. No issues. 
However, with exactly the same configuration, and just by enabling Quick_Home, the motors hit home position and start grinding for a couple of seconds, and if sensitivity was too low.  
If sensitivity value is increased, I start to have false positives. 

### My Configurations
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4527904/configs.zip)

### Steps to Reproduce
1. Disable quick_home, and home printer: no issues.
2. Enable quick_home, and home printer: motor grind in home position. 

**Expected behavior:** Sensorless homing should behave similarly with  and without quick_home enabled. 

**Actual behavior:** as described above. 

#### Additional Information

- Platform: Ender3
- board: SKR v1.4 Turbo
- drivers: TMC2209
- Sensorless homing in X and Y, limitswith on Z.

craigistired, 2020-04-25:
I had the same issue. With Quick_Home enabled the sensitivity is always either too high (fails to home) to low (grinding if already homed). Disabling Quick_Home resolved this issue.

thinkyhead, 2020-05-14:
There are various remedies you can try, such as the sensorless homing backoff, but this isn't strictly a bug. It's just a limitation of quick homing combined with sensorless homing, especially in noisy electronic environments on the less expensive boards. If no amount of tuning helps, you may just have to disable it.

thinkyhead, 2020-05-14:
Hmm… On review it does appear our trinamic code is lacking some corexy consideration. Please give #17996 a try and see if it helps.

closed 2020-05-14 by thinkyhead

craigistired, 2020-05-24:
@thinkyhead Homing backoff is a good solution to the grinding when homed issue with sensorless x/y homing.