MarlinFirmware/Marlin #17693: [BUG] Marlin 2.0.5.3 only heats up to HEATER_0_MAXTEMP minus 15 Degrees

JoergGu, 2020-04-24:
### Bug Description

Marlin 2.0.5.3 only heats up to HEATER_0_MAXTEMP minus 15 Degrees

### My Configurations

#define HEATER_0_MAXTEMP 260

### Steps to Reproduce

Selecting temperature from Menu an try to set it to max. Temp or just trying to print objects within the temperure range of max. temp and max. temp minus 15

**Expected behavior:** possibility to set temperature up to max. temp.

**Actual behavior:** maximum is max. temp. minus 15

#### Additional Information

Found entries in temperature.h and temperature.cpp

temperature.h, line 640:
temp_hotend[ee].target = _MIN(celsius, temp_range[ee].maxtemp - 15);

temperature.cpp, line 411:
if (target > GHV(BED_MAXTEMP - 10, temp_range[heater].maxtemp - 15)) {

Probably forgetten entries used for testing?

thisiskeithb, 2020-04-24:
It’s a safety feature. If your printer hits max temp, Marlin will halt because of thermal runaway protection.

If you regularly need to print near/at the desired temp, increases your `*_MAXTEMP` settings (assuming you’re printer can safely reach/print at those temps).

boelle, 2020-06-20:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

thisiskeithb, 2020-06-20:
> Configurations, please
> Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
> requested in the Issue template) and drop them into your next reply.
> We'll check them over and see if anything is amiss.

@boelle: Configs won't help here since it's an intended safety feature of Marlin to only allow the user to set the hotend temp 15 degrees below `HEATER_0_MAXTEMP` and 10 degrees below `BED_MAXTEMP`.

The idea being that if you hit the max temp, something is wrong & Marlin halts your printer.

boelle, 2020-06-20:
so actually there is not a bug and we can close it?

sorry for the many canned replies tonight but we have many issues that have gone stale etc and we need to get through them

only me and @sjasonsmith doing that as far i know

thisiskeithb, 2020-06-20:
It’s a safety feature of Marlin, so the issue can be closed.

closed 2020-06-20 by boelle