MarlinFirmware/Marlin #17696: [BUG] no text display for change filament

88mich, 2020-04-24:
### Bug Description

no item is displayed in the lcd menu under the item change filament.
when the English language is selected I can choose the extruder to change the filament but if I select the Italian language nothing appears in the menu

### Steps to Reproduce

My printer is a A10M so i think the problem is in the italian's multi extruder menu

1. set multiextruder print such as A10M
2. change language to italian
3. go in the print menu > cambia filamento>

I think the problem is because in Italian there are too many characters for that voice

GMagician, 2020-04-25:
I can't check now (printer is busy) but if messages are too long you may try to shrink them and test.
`filamento` may be written as `filam.`.
All messages are within 20 char, but in some menù are required max 17 chars.
Usually I translate all messages but sometimes I don't have options installed or I misunderstand their meaning and translation is not perfect.
If you find some "bad" translation or other "too long" messages and you're not able to post a PR, let me know

88mich, 2020-04-25:
[image]

I confirm, there are too much character for thoose Lines. I change "filamento" with "filo" and everything works fine

GMagician, 2020-04-25:
That's good will you post a PR for it or should I manage it for you?

GMagician, 2020-04-25:
it's better 'filam.' or 'filo'? Correct translation should be the first in my opinion

GMagician, 2020-04-25:
Probably also `MSG_FILAMENTUNLOAD_E` should be shrinked to `Rimuovi filam. *` or `Rimuovi filo *`

If 'filo' is used I think all 'filamento' references should be changed too, to be consistent

88mich, 2020-04-25:
If you manager It for me would you do me a favor.
  i think it's ok filam.  or filo.  I wrote filo only because I had already seen it written in other voices

Thank you

A: MarlinFirmware/Marlin
Cc: 88mich; Author

GMagician, 2020-04-25:
@88mich you may close this now...fix has been merged

closed 2020-04-25 by 88mich