MarlinFirmware/Marlin #17699: [BUG] LCD screen issues with ubl and scrolling

phazei, 2020-04-25:
SKR 1.4 turbo / TMC2209
bugfix 4/23
BTT - Full Graphic Smart Controller

https://imgur.com/a/ZgANwsP
This is the only place this happens, whenever it's on this mesh editing screen from ubl, only the bottom half of the text is visible

There's also this issue that happens when adjusting any value that changes quickly with the scroll wheel:
https://photos.app.goo.gl/4W3nuJWyDXNtfck2A

DerBierBaron, 2020-05-14:
I have the same issue in the mesh editing screen, but with the stable build 2.0.5.3. 

It occurs when the mesh is to be edited manually, but for me its both when I start from the screen or and while I'm sending G29 P4 T via pronterface.

My setup:
SKR v1.3 /Watterott TMC2209
BTT Full graphic smart controller

boelle, 2020-06-20:
@phazei still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle