MarlinFirmware/Marlin #17705: Support for emergency parser for STM32

jimhiggs, 2020-04-25:
I have two issues that I believe require emergency parser to be enabled for my printer based on BTT SKR Pro V1.1.
1 I cannot remotely stop the machine from ESP3D while heating (waiting for temperature).
2 Unable to implement filament runout sensors which require advanced pause which requires emergency parser.
There is an alternative to number 2 connecting filament sensor to the BTT TFT3.5 But I had wired my 3 filament sensors to 3 ports on the SKR Pro and defined each separately. If using the TFT there is only provision for one so would have to wire in serial and treat as one.

Is there likely to be support for emergency parser for STM32 any time soon or should I go the TFT route?

thisiskeithb, 2020-04-26:
> Unable to implement filament runout sensors which require advanced pause which requires emergency parser.

Which filament sensors are you using? Filament runout is supported on `STM32`.

jimhiggs, 2020-04-26:
I am just using simple microswitch type. Problem is  if I enable runout sensors I also have to enable Advanced Pause Feature or build fails. If I enable Advanced Pause Feature, I have to enable Emergency Parser or build fails. Note I just tried again and the error says "ADVANCED_PAUSE_FEATURE currently requires an LCD controller orEMERGENCY_PARSER". 
In the past I had LCD Defined but it then said "EMERGENCY_PARSER Required." not sure if that is a recent change in Marlin. But I have a BTT TFT3.5 not an LCD and I use ESP3D. 
I was hoping I could  add a custom command on the TFT for the continue etc or handle them from ESP3D. I know I can wire the sensors to the TFT but would prefer it to be handled in Marlin.
It is Emergency Parser that is not supported on STM32.

thisiskeithb, 2020-04-26:
> It is Emergency Parser that is not supported on STM32.

Correct, but as the error message you pasted above says, `ADVANCED_PAUSE_FEATURE` doesn’t require `EMERGENCY_PARSER` if you have an LCD controller.

I use/support filament runout on several `STM32`-based builds, but they have some kind of 2004/12864 LCD controller or dual-mode BigTreeTech TFT. Side note: BigTreeTech’s firmware doesn’t know what to do with `M600`/filament change currently, so you’d be better off plugging your sensor directly to the TFT.

Having said all that, it would be nice if `EMERGENCY_PARSER` was supported on `STM32` 👍.

jimhiggs, 2020-04-26:
It also makes me a little bit worried when I can't kill remotely during heat up.  I understand Emergency parser would fix this?

tphan26363, 2020-04-26:
AND SADLY THE stm32  come from SKR BTT board which also come with TFT-35 or TFT24.  These controller are not a compatible LCD  and has no setting in configuration to turn on the LCD.  That is why the STM filament run out on STM32 is dead code

chestwood96, 2020-05-11:
Does anybody know what actually prevents the emergency parser from working on stm32?
Is it because of the emulated/simulated serial port it uses or is there something different missing?

I tried force enabling it by disabling the sanity check and at the very least it did not blow up but also did not work (I kind of expected it to crash or something).

I would really like to get this working so I can cancel user inputs over octoprint without having to use the main power relay.

Could someone point me in the right direction?

tphan26363, 2020-05-14:
I think from my point of view( I did messing with  the planner), M108 can not be insert into lower level code when it is running in a fast loop of the core STM32. Most likely when it doing this you will see a communication timeout.  probably because  the STM32 was running too fast.  The core code was not writing to support a real time clock but to run at what ever clock speed is.  (I am thinking of pacman).  model code should support code running at multiple divided clock speed to not hog all of the resource just waiting for a simple OK from user

boelle, 2020-06-20:
@jimhiggs still an issue?

jimhiggs, 2020-06-21:
Looks like the Emergency arser support issue is being handled in #18095.

closed 2020-06-21 by boelle