MarlinFirmware/Marlin #17709: [BUG] Virtual Tool T1 is set to 0:0%

iatwalrus, 2020-04-25:
When attempting to print **not** using virtual extruders in GCode, but using tool change references (T0, T1, etc.), the printer will lockup when selecting T1 as the extruder to print from. I noticed in the Mixer menu, if you set Active V-Tool to 1, the mix is set to 0:0%, and this needs to default to 0:100%. It can be manually adjusted and saved, but the default should be edited. All other tools have an appropriate default percentage. This way when the firmware is installed, manual configuration of this setting is not required and it will work using tool references in GCode instead of virtual

firepower9966, 2020-04-29:
I also can confirm too.

lterrill, 2020-04-30:
Duplicate of #17635, patched in bugfix-2.0.x branch

closed 2020-06-20 by boelle