MarlinFirmware/Marlin #17710: z-stop not working at second probe point

Thijsvanzwam, 2020-04-25:
I have an issue on a BTT SKR 1,4 using ABL. 
The z-stop is connected to an inductive probe so I tried swapping it for a switch but get the same error. When activating ABL using G29 or on the screen (BTT TFT3,5 v3) it probes the first point which all goes well but when it goes to the second probing point it doesn't respond to the z-stop anymore, giving an error.
z-stop works fine on autohome so the connection on the board is ok.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4534000/config.zip)

tried Marlin 2.0.5.3 and bugfix

Thijsvanzwam, 2020-04-26:
Did some more trying but no luck... Homing works but leveling gives an error. The z-motors keep spinning and ignore the z-endstop. (used a switch for testing now) 

config files:
[Archief.zip](https://github.com/MarlinFirmware/Marlin/files/4535121/Archief.zip)

boelle, 2020-06-20:
@Thijsvanzwam is the issue still there?

Thijsvanzwam, 2020-06-21:
Yep, still there. Went back to Marlin 2.0.0 where the problem is not there. can't figure out what the reason is.

shitcreek, 2020-06-23:
I've running the same board on one of our machines and using `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` without any issue. Have you tried that? Which pin on the board is your probe connected to?

thinkyhead, 2020-06-23:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thinkyhead, 2020-06-23:
> The z-stop is connected to an inductive probe so I tried swapping it for a switch

Inductive probes and switches may differ in the triggered state, so you need to use `M119` to confirm that the endstop inverting is correctly set. It also helps to make sure the pull-up / pull-down is set where needed, because floating pins will return random values.

thinkyhead, 2020-06-23:
It is unclear that there is a bug, and we can't do safaris. So I need to drop this here…

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-06-23 by thinkyhead