MarlinFirmware/Marlin #17711: [FR] Send to SD card and print at the same time

jan-gerard, 2020-04-25:
### Description

As writing to an SD card via the printer interface is extremely slow (similar to printing), it would be advantageous to have an option to do both at the same time: print and write all gcode to an SD card file. This would be very useful if you want to print one design multiple times.

### Feature Workflow

1. Send a gcode command to specify the filename to save the data to.
2. Send a gcode command that tells Marlin to print and save to file at the same time.
3. Print your object.
4. Print it again and again from SD card.

#### Additional Information

ErwinRieger, 2020-04-26:
FYI: Did this some years ago in (my fork of) ultimaker's marlin fork (https://github.com/ErwinRieger/Ultimaker2Marlin-USBPrint, no longer active).

The key point was to open the file on the sdcard in *O_RDWR* mode. This way the file could be written through usb and read by the planner at the same time. This way the file on sdcard acts as a (big) buffer for gcode commands (something like a swap file). Give the gcode file some head start and then begin printing it, this works as long the average download rate is greater or equal the read-rate of the planner.

Some documentation: https://github.com/ErwinRieger/Ultimaker2Marlin-USBPrint/wiki#sd-buffer-file-in-o-rdwr-mode

Code CardReader::openFile() where file is opened in RDWR mode: https://github.com/ErwinRieger/Ultimaker2Marlin-USBPrint/blob/master/Marlin/cardreader.cpp#L346

rhapsodyv, 2020-11-22:
I think there’re better ways to solve this. 

Most of the 32 bits boards will be able to use serial and SD at up to 4Mb/s. 
And we have binary file transfer, that uses compression to transmit the file, so it’s even faster.

And if you are doing a lot of prints, you can just copy the file to the sd on the computer. Or even just print it over serial over and over again. You would need to remove the piece the printer between prints.